motioneye-project/motioneyeos #308: Idea: network sniffing for devices for auto-on/off instead of schedule

petermolnar, 2016-05-24:
I have an idea, but being entirely new to motioneyeos, I don't know where to start implementing it. 
In short: a tcpdump sniffing or querying the router for connected devices could reveal what MAC addresses are present on the network. This would be useful to automatically turn the camera on or off depending if certain devices (mobile phones, mostly) are present on the network or not. This way there might be no need for schedules for some.

So the questions: are they any ways to script motion detection, recording and camera states from the command line?

closed 2016-05-24 by petermolnar

felixgr, 2016-08-10:
@petermolnar - did you work on this? I plan to work on this.

@ccrisan any comments from your side?

ccrisan, 2016-08-10:
Nobody is working on this as far as I know, given the fact that motionEye currently lacks a way to programmatically control motion detection and I'd need to implement it myself.

felixgr, 2016-08-10:
Thanks for the reply. Indeed, if there is no command line utility, that makes such a feature harder to implement. I might give it a try and report back in case of success.