motioneye-project/motioneyeos #317: motioneyeos-raspberrypi2-20160410 image will not boot

gylhen, 2016-05-31:
I have tried to boot from both my rbp-2 and -3 with their resp. 20160410 images but there is just a brief "green light" and then nothing (network connected)?!

ccrisan, 2016-06-01:
https://github.com/ccrisan/motioneyeos/wiki/Troubleshooting#system-wont-boot

gylhen, 2016-06-01:
Thanks,
I have verified all this before opening this issue:
*yes, for both rbp2&3 - you have used the OS image that is suitable for your board

*of course! (also checked content of card after writing) - you have extracted the gzipped OS image before writing it to the card

*yes. used it for other working images - you use an SD card that is compatible with your board

*yes, checked that as well - the system is indeed not booting by connecting a monitor and observing the output

Before reporting an issue, you should:
*well, the card worked with other images on it both before and after writing this image so seems a bit overkill to buy another one! - try using another SD card (preferably a different model)

*yes, as stated above I have done that in several ways an they all work - try "burning" images of other OSes that normally work on your device and see if that works

ccrisan, 2016-06-01:
Oups, then I'm out of ideas. Does at least the rainbow appear on the monitor?

gylhen, 2016-06-01:
Nope, it is all black (there is only a short green blink on the rbp)
Strange!

gylhen, 2016-06-01:
Just had another go and it seems that my hdmi-connection had some problems: I now do get the boot sequence on the monitor!
However, there seem to be a problem when configuring the network connection and ends after a while and reboots (looping).

ccrisan, 2016-06-01:
Well in that case you'll need to fix the network connectivity issues. This might help you https://github.com/ccrisan/motioneyeos/wiki/Troubleshooting#system-continuously-rebooting.

closed 2016-06-01 by ccrisan

gylhen, 2016-06-01:
IT WORKS!
After some struggling with my Router I got it up and running now!
Thanks for your support! And a Great Product!