motioneye-project/motioneyeos #319: No local cam on motioneye

Galaxyflyer, 2016-05-31:
Because of issue #318 I installed motioneye on raspberry jessie, motioneye works but doesn't find the onboard camera. I can find remote raspberry cameras but not the local one, reinstalled modprobe bcm2835-v4l2, no joy.... The onboard cam works fine through Raspicam though... any ideas?

Other than the last issues, excellent program, been trying it on 2 different Pi's for about a week now, runs smooth! Thanks!

ccrisan, 2016-06-01:
What does `ls /dev/video0` say?

Galaxyflyer, 2016-06-01:
Will need to get back to you on that, just reinstalled motioneyeOs on the wrong SD card...Oops...

Galaxyflyer, 2016-06-01:
Was the correct Sd after all..

the /dev/video0 is non existant...

ccrisan, 2016-06-01:
Then make sure the module bcm2835-v4l2 is loaded.

Galaxyflyer, 2016-06-01:
Loaded and it works now, Thanks! Do I need to load it again after every reboot?

ccrisan, 2016-06-01:
just add it to `/etc/modules`.

closed 2016-06-01 by ccrisan

jatpatel1, 2016-07-23:
Hi, Trying to get this working on pi2, camera v2.1 and same issue with camera not found.
Really new to this so appreciate the help.
How do I load the `bcm2835-v4l2` module?
When I do a grep(`less /proc/modules |grep bcm`) on the module list, it's not there. Do I need to install it first. Also I have made sure that the camera is enabled.
Thanks

xhain, 2017-08-13:
Same issue as jatpatel1 here.
Running MotioneyeOS on a Raspberry Pi 2 with the v2 Camera module.
If I try to add a local camera it says 'no camera'.

Any idea what how I could fix this?

ccrisan, 2017-08-14:
@xhain what version of motionEyeOS? Does it work with Raspbian?

xhain, 2017-08-14:
@ccrisan I realized I'm an idiot and plugged the camera into the display slot 🙄