motioneye-project/motioneyeos #320: No wifi on motioneye Pi 3

Galaxyflyer, 2016-05-31:
I just bought a Pi3, installed the latest motioneye OS and configured the wifi details. When unplugging the cable and rebooting, motioneye doesn't detect the wifimodule of the Pi3, so it loops into reboot.

ccrisan, 2016-06-01:
Can I see your boot.log file?

Galaxyflyer, 2016-06-01:
[boot.log.txt](https://github.com/ccrisan/motioneyeos/files/292944/boot.log.txt)

ccrisan, 2016-06-01:
Are you absolutely sure your device is a Raspberry PI 3 and the OS image you have downloaded is the Raspberry PI 3 version?

Galaxyflyer, 2016-06-01:
It is a raspberry Pi3... I might have used the same version as the on my Pi2's... I'll reinstall and let you know... thanks! Any ideas on the #319 issue?

Galaxyflyer, 2016-06-01:
Problem solved! Thanks for the swift reply!

closed 2016-06-01 by ccrisan