motioneye-project/motioneyeos #321: Any plan to port this OS to the Pine64 board?

havensal, 2016-06-03:
I've been trying to get this running on my Pi0.  I read that a fix is in the works.  I am curious if you have access to a Pine64 board and if you plan on working on a version of this OS for that board.  I ordered one through the Kickstarter campaign.  Thanks.

ccrisan, 2016-06-03:
If I get my hands on one of those, I'll do my best to port motionEyeOS to it.

DrCord, 2016-09-10:
I may end up with an extra, I could potentially donate it to you for this effort.

ccrisan, 2016-09-10:
If you send one of those to my address I will do my best to port the OS to it, although I'm not promising anything before getting my hands on the board and see what it's capable of.

DrCord, 2016-09-21:
it seems like as motioneye runs fine under linux that porting it to Pine64 is a waste of your time...?

ccrisan, 2016-09-21:
Hmm is that a question? Porting motionEyeOS to a new and relatively popular board should never be a waste of time.

DrCord, 2016-09-21:
ok, cool, I didn't want to waste your time with something no one would use...If/when I get the other Pine64 I am due I will probably send one your way then.

DrCord, 2016-10-19:
It looks like they are failing to send me my second board so this probably won't happen.

CmbtDocThor, 2017-01-01:
@Ccrisan any updates on this? Seems like it would be a good use for the pine64. I bought one, and am a curious beginner.

DrCord, 2017-01-01:
> Nope. I haven't gotten one yet. I've been busting my head trying to port
> motionEyeOS to Orange PI and it's been a total failure so far.

I have a pine64 but was hoping to sell it instead of giving it away...

CmbtDocThor, 2017-01-01:
Too bad, it would be good use for it, in my opinion. Seen a couple youtube videos of linux running smoothly on the Pine64. Maybe we should crowdfund you for the project, an incentive to make a port/prototype , lol.

DrCord, 2017-01-01:
if anyone wanted to slightly incentivize me I would totally be willing to send this in...

rluncasu, 2017-02-27:
@ccrisan I got a pine64+ 2gb available, coincidentally I'm in Bucharest so if you haven't left the country yet ;) , just send me your address and I'll send it to you.
Alternatively , you could guide me into porting motioneyeos myself.

ccrisan, 2017-02-27:
@rluncasu no, I haven't left the motherland yet :) I would rather do my best and guide you through porting the OS than having another board on my desk. However once we have it running on the new board, I'd expect you to help with testing new releases. Would you be up to the task?

rluncasu, 2017-02-28:
@ccrisan ok, up for it. Let's go, what is step 1 ? :)

ccrisan, 2017-03-02:
I'm trying to separate the motionEye-specific part of the OS from the OS itself. Therefore I created a new proect/repo called [thingOS](https://github.com/ccrisan/thingos). It's basically the OS without all motion(Eye) related packages. The reason behind this is to be able to reuse all the work that has been done for the OS itself for other similar projects, mainly from the IoT world (hence the name thingOS).

I'm telling you this because the next release will be based off thingOS and porting motionEyeOS to a new board mainly consists in making thingOS run on the new board.

Also let's take this to our new [google group](https://groups.google.com/forum/#!forum/motioneye). I'll write a post with some basic steps required to port the OS to another platform.

ccrisan, 2017-06-01:
Actually we should keep this open as the port to Pine64 is in the works. I got my Pine64 board and I'll try to do the port.

bortek, 2017-06-01:
Sounds good. My board arrived couple of weeks ago. Id be happy to test the image when it's ready

ccrisan, 2017-06-11:
https://github.com/ccrisan/motioneyeos/commit/bf37f786c02646b28908bfed8db3f0fe5104fbcd adds support for Pine64.

closed 2017-06-11 by ccrisan