motioneye-project/motioneyeos #323: Email and drive/ dropbox issues

graphicted, 2016-06-04:
Hello seems like i'm unable to receive email notifications and unable to upload the motion movies to Dropbox or Google drive.

checked in the motioneye.log and saw this error 🎱 

2016-06-04 13:33:34: [sendmail]    ERROR: failed to send mail: timed out
Traceback (most recent call last):
  File "/usr/lib/python2.7/site-packages/motioneye/sendmail.py", line 208, in on_message
  File "/usr/lib/python2.7/site-packages/motioneye/sendmail.py", line 52, in send_mail
  File "/usr/lib/python2.7/smtplib.py", line 256, in __init__
  File "/usr/lib/python2.7/smtplib.py", line 316, in connect
  File "/usr/lib/python2.7/smtplib.py", line 291, in _get_socket
  File "/usr/lib/python2.7/socket.py", line 575, in create_connection
timeout: timed out

No idea which is the log for the network sharing.
thanx a lot!

Gordon-1, 2016-06-06:
I don't know if this is related but I just upgraded to 20160410 and discovered that M-OS stopped sending email.. I found that the contents of the fields for the email settings were shifted one place.  I re-entered my email settings and it is OK now.

ccrisan, 2016-06-06:
There were some changes to the email parameters but I had hoped they were backwards compatible (it turns out they are not).

barkingdoggy, 2016-06-10:
I'm running 20160605.  Here's what I'm getting trying to email to smtp.gmail.com using TLS on port 587...

2016-06-10 17:34:35: [sendmail]    ERROR: failed to send mail: [Errno -2] Name or s
Traceback (most recent call last):
  File "/usr/lib/python2.7/site-packages/motioneye/sendmail.py", line 208, in on_me
  File "/usr/lib/python2.7/site-packages/motioneye/sendmail.py", line 52, in send_m
  File "/usr/lib/python2.7/smtplib.py", line 256, in __init__
  File "/usr/lib/python2.7/smtplib.py", line 316, in connect
  File "/usr/lib/python2.7/smtplib.py", line 291, in _get_socket
  File "/usr/lib/python2.7/socket.py", line 557, in create_connection
gaierror: [Errno -2] Name or service not known

ccrisan, 2016-06-11:
Your DNS server is misconfigured.

barkingdoggy, 2016-06-13:
I don't think so.  When I ssh to the device I can ping google.com
successfully.

ccrisan, 2016-06-13:
Are you sure your SMTP server is correctly spelled (smtp.gmail.com)? If yes, can you ping smtp.gmail.com from via SSH?

barkingdoggy, 2016-06-13:
D'oh!
I had smpt

ccrisan, 2016-06-13:
I'm glad you found the problem. I'll close the issue now.

closed 2016-06-13 by ccrisan

barkingdoggy, 2016-06-13:
Sorry to bother you.  Thank you for helping me see the problem.  Thank you
for your good work on Motioneye!