motioneye-project/motioneyeos #324: Google Drive / Dropbox

adriangibbons, 2016-06-05:
Hi, the project has come a long way since MotionPi - thank you very much for making this available. The front-end is great!

I just can't get much luck with either Google Drive or Dropbox. I have tried both, obtained key, entered folder name, tested and it reports success. If I spell the folder name wrong and test it says fail. So I have some confidence that something is working...

However none of the files ever find their way to either Google Drive or Dropbox. For testing I started capturing still images using motion-detection, but moved to interval of 30 seconds so I didn't have to keep waving at my camera! :)

I attach log files (changed extensions so they'd upload). I am using a Raspberry Pi B+

Have looked at other's issues but can't work-out what's going on. I tried formatting and re-writing the image to the SD Card for a second time but didn't make any difference.

[messages.txt](https://github.com/ccrisan/motioneyeos/files/299230/messages.txt)
[motion.txt](https://github.com/ccrisan/motioneyeos/files/299231/motion.txt)
[motioneye.txt](https://github.com/ccrisan/motioneyeos/files/299232/motioneye.txt)

ccrisan, 2016-06-05:
@adriangibbons do you use any special characters in your admin password? If yes, could you try setting a simple password and see if the problem persists?

adriangibbons, 2016-06-06:
Hi, yes I did - I had an ! in my password. I have removed it and now both Google Drive and Dropbox work :))

Dropbox creates a folder with the date; however it doesn't look like the Google Drive interface is doing the same.

Is that by design? I guess that is why when selecting Dropbox as the upload service you get a "Include Subfolders" option?

Thanks for your quick response! :+1:

ccrisan, 2016-06-06:
The new version should accept special characters in your password. The Google Drive/Dropbox feature is still rough around the edges and needs a lot of improvements, I must admit that.

closed 2016-06-06 by ccrisan

adriangibbons, 2016-06-07:
No dramas - it's a pretty important feature for me as I use a Raspberry Pi and it would be quite easy for an intruder to just pick it up and walk off with the photos.

I'm going to switch to using Google Drive as I have a lot more storage space there. As a workaround, instead of naming files with a folder i.e. "YYYY-MM-DD **_/_** HH-MM-SS" I will just incorporate the date into the file name instead of having them in subfolders i.e. "YYYY-MM-DD-HH-MM-SS".

Thanks again for access to this great project! Adrian

rfv-370, 2016-12-09:
Hi Crisan,
I am strugling a bit with the dropbox connection. Just to confirm, the path is the one that is provided by dropbix when sharing a folder or I copy the path from my browser ?
I tried both and requested keys but I end up with connection error on both... any ideas of suggestion?
best
Robert

ccrisan, 2016-12-09:
Yes, path should be the dropbox path (e.g. `mycam` or `/mycam/subdir1`). The path must exist on dropbox first.

rfv-370, 2016-12-10:
Hi Crisan,
thanks, I have path like:
/home/Neva_Cerbere/Cerbre_Chtly
which does exist on my dropbox. may be removing the '/home' will do it..?

rfv-370, 2016-12-10:
and I realise I am on the wrong thread... as I use MotionEye (and not motionEyeOS)...

I have done another test with only "/Neva_Cerbere/Cerbre_Chtly" and this time I got the "Service connected". I applied. doing some more test now to check if the upload works.

rfv-370, 2016-12-10:
ooooohh Happy Days! 
:-)
That works now!
Thanks Crisan for you good job on this project.