motioneye-project/motioneyeos #328: multithreaded performance

AdamMalkin, 2016-06-06:
Hi,

I am building a home surveillance system using 3-4 pi zeros and B+ board and the pi cameras.
These will be running motionEyeOS as simple network cameras and i will use your "Multiple devices with a central server" scenario.

My question is, is it worth me getting a Pi 3 for the "central server" or should i stick with my pi 2? how well does motionEyeOs spread the image processing over multi core processors?

Thanks,
Adam

ccrisan, 2016-06-06:
The CPUs will be used as efficiently as a Linux machine can. I use an Odroid XU4 with 8 cores and it works like a charm with 9 cameras (system load below 4 on average). I never used a RPI2 or a RPI3 as a central server so I wouldn't know if it's worth the difference.

AdamMalkin, 2016-06-06:
I will give it a go and report back, thanks for the speedy reply

closed 2017-06-01 by bortek

HJSK, 2020-04-12:
Hi ccrisian,
First great thanks to your fantastic job!  
Which CAM's you use ccrisan? CSI, IP mjpeg, 264 etc..
I notice a big difference in cpu load at csi vs ip cam's. 
Also on the rpi 4 which should be 3-4 times faster then rpi3 
Both rpi's are very quickly reaching the 80 degrees border,
unless I use cases from aluminium so big cooling

ccrisan, 2020-04-12:
> Which CAM's you use ccrisan? CSI, IP mjpeg, 264 etc..

I use Raspberry Pi 5MP CSI cameras in FNC mode.

> I notice a big difference in cpu load at csi vs ip cam's.
> Also on the rpi 4 which should be 3-4 times faster then rpi3
> Both rpi's are very quickly reaching the 80 degrees border,
> unless I use cases from aluminium so big cooling

You shouldn't run a RPi4 without cooling.