motioneye-project/motioneyeos #333: Install on Debian

arkhan, 2016-06-08:
Hello,

Its posible to install MotioEyeOs on Debian Server

Thanks

boubou30, 2016-06-09:
Hello, 

## Debian 8 : Installation motion + motioneye

1:  echo "deb http://www.deb-multimedia.org jessie main non-free" >> /etc/apt/sources.list
2:  apt-get update
3:  apt-get install deb-multimedia-keyring
4:  apt-get update
5:  apt-get install motion ffmpeg v4l-utils
6:  apt-get install python-pip python-dev curl libssl-dev libcurl4-openssl-dev libjpeg-dev
7:  pip install motioneye
8:  mkdir -p /etc/motioneye
9:  cp /usr/local/share/motioneye/extra/motioneye.conf.sample /etc/motioneye/motioneye.conf
10: mkdir -p /var/lib/motioneye
11: cp /usr/local/share/motioneye/extra/motioneye.systemd-unit-local /etc/systemd/system/motioneye.service
12: systemctl daemon-reload
13: systemctl enable motioneye
14: systemctl start motioneye
15: pip install motioneye --upgrade
16: systemctl restart motioneye

look at the configuration file in "  /etc/motioneye/motioneye.conf "   on the line " the TCP port to listen "  to log  the virtual interface. for me it was http://www.example.com:8765.

++

arkhan, 2016-06-14:
cool!!!
Thanks

closed 2016-06-14 by arkhan