motioneye-project/motioneyeos #335: Rainbow Screen on Raspberry 2, desbian works

fettsvett201, 2016-06-10:
Hello

I am getting the rainbow screen when trying to boot from the Raspberry 2

I am using the latest version (motioneyeos-raspberrypi2-20160605).

I have used raspbian successful with the microsd card (both before and after trying to use motioneyeos)

I have also tried "boot_delay=1" with no success. 

The only items connected is the ethernet cable, the raspberry pi camera v2, hdmi cable, and the power cord. The ethernet port works as I have tested it with raspbian.

Any thoughts?

fettsvett201, 2016-06-10:
The kernal has the following MD5 address (if it matters) = BBB01C4414FFEC175617C29F88FB9A6A

ccrisan, 2016-06-10:
https://github.com/ccrisan/motioneyeos/wiki/Troubleshooting#system-wont-boot

fettsvett201, 2016-06-13:
I have tried three different cards (Lexar, Sandisk, and PNY).
All three run raspbian successfully.  All three were able to turn on the camera and take pictures. All three were able to connect to the Internet via the ethernet cable.

fettsvett201, 2016-06-14:
Please excuse my ignorance. I haven't been in the pi game for long and didn't realize model 1 B looks EXACTLY the same as Model 2 B when its in an enclosure. I thought for sure I ordered a 2 but apparently I don't know what Im doing.

Im closing the issue. Thank you for trying to help.

closed 2016-06-14 by fettsvett201