motioneye-project/motioneyeos #336: SSH access stopped working (bad password) - but web access still works.

tajohnsen, 2016-06-10:
I had an odd issue where I could log in as admin via the web but I could not SSH.  Rebooting didn't help.  I reset the password on the web interface (using the same password) and then I could SSH.  The stored password in my web browser still worked after the reset so it isn't like I memorized it wrong.  

What triggered me to SSH was when I noticed that the camera LED was going off and on about every 10 seconds; on the web interface the camera was turning on and off also.
My /var/log/messages is full of:
`Dec 31 19:00:04 SecCam auth.err passwd: can't update password file /etc/passwd                                                                    
Dec 31 19:00:04 SecCam auth.err passwd: no record of admin in /etc/shadow, using /etc/passwd`
I've seen these on other people's logs so it might be something going on in the boot process.  

An couple odd lines in the log popped around the date this started happening. And no, there isn't a typo with the date.
`Jun 10 Dec 31 19:00:04 SecCam auth.err passwd: no record of root in /etc/shadow, using /etc/passwd                                                
Dec 31 19:00:04 SecCam auth.err passwd: no record of admin in /etc/shadow, using /etc/passwd`

The problem seems to be resolved but I'm still curious.  I can't see anything like this in the forums.  Any ideas what could cause such a problem?
Tim

ccrisan, 2016-06-12:
It might have been a problem with the very first boot of your system, where things are configured for the first time. These things include the root/admin account.

closed 2016-07-03 by ccrisan

tajohnsen, 2016-07-04:
Hi, sorry I didn't reply sooner. I have the camera up on a wall that's beat down by the sun. On hot days it overheats and that's when this issue starts. I can't pinpoint why SSH stops working but it's one of many symptoms. Hope this helps.

Chiny91, 2016-07-04:
I get the same problem from time-to-time, including at the moment.   Ingenious idea to change the password to itself but sadly that has not worked for me.

ccrisan, 2016-07-04:
@Chiny91 clean install? If yes, can I see some log files? The SSH daemon does not just die without a cause :)

Chiny91, 2016-07-04:
Clean install of _Current Version 20160612_, compiled with latest Debian Jessie of that date, including _rsync_ (worked well) and _ntpq_ (I'm an ntpd addict).  (Probably) set running on 28/6/16 and the cameras are still displaying but no stills stored since 1155pm on 30/6/16.   The ssh daemon appears to be running as I get a response to a connect request with the user (admin) accepted, but the password is refused.

I just tried restoring the backup config but that failed (no reason given).

The log stopped mid-line on 30/6/16.
`
[motioneye.log.txt](https://github.com/ccrisan/motioneyeos/files/346063/motioneye.log.txt)
`

ccrisan, 2016-07-04:
So you have compiled the OS yourself. It's hard to tell what's going on without having a look over the OS image (either running or as a backup of the whole SD card). In any case, judging from your logs, things look terribly wrong.

Chiny91, 2016-07-04:
Ha yes, it looks totally broken to me.    Time to d/l the latest, re-compile and break that one, too 😇

ccrisan, 2016-07-04:
Well I must admit I never tried compiling everything from scratch on a brand-new machine. I always use a Debian machine for that and the same old directory :)

Chiny91, 2016-07-04:
For extra fun, I'm using a _virtual_ Debian machine, running in VirtualBox on an iMac.     motioneyeos is compiling as I type.

Which reminds me... the _Building From Source_ Wiki page gives an optional line of:
`./build.sh [board] xconfig`
I don't have X server running on my Debian as I ssh in from OSX on my iMac.   So, I use:
`./build.sh [board] menuconfig`
which looks nice enough.
Hardcore terminal folk might use:
`./build.sh [board] config`
Just a thought for the excellent Wiki.

ccrisan, 2016-07-04:
Yeap, menuconfig could be used in non-X environment, although I couldn't get by w/o the xconfig frontend these days :) The search function in menuconfig is just... well... useless, most of the times. Anyways, fingers crossed and let's see if the build succeeds and works as expected this time.

Chiny91, 2016-07-04:
Yes, all running properly again with the current code freshly compiled, complete with ssh login.
`Jul  4 15:43:57 bananapi auth.info sshd[4720]: Accepted password for admin from <local IP> port 61224 ssh2`
Fingers crossed :persevere: