motioneye-project/motioneyeos #337: Cannot get Email notifications

1966tim, 2016-06-11:
I don't know what I am doing wrong 
Email Addresses	
my.email@gmail.com
SMTP Server	
smtp@gmail.com
SMTP Port	
587
SMTP Account	
my.email@gmail.com
SMTP Password	
•••••••••
From Address	

?
Use TLS	ON

1966tim, 2016-06-11:
ERROR: failed to send mail: [Errno -2] Name or service not known
Traceback (most recent call last):
  File "/usr/lib/python2.7/site-packages/motioneye/sendmail.py", line 208, in on_message
  File "/usr/lib/python2.7/site-packages/motioneye/sendmail.py", line 52, in send_mail
  File "/usr/lib/python2.7/smtplib.py", line 256, in __init__
  File "/usr/lib/python2.7/smtplib.py", line 316, in connect
  File "/usr/lib/python2.7/smtplib.py", line 291, in _get_socket
  File "/usr/lib/python2.7/socket.py", line 557, in create_connection
gaierror: [Errno -2] Name or service not known

ccrisan, 2016-06-11:
Your DNS server is misconfigured.

1966tim, 2016-06-11:
Oh, now I have no idea of what to do about that. Sigh. Thanks for the reply.

1966tim, 2016-06-12:
I edited /etc/resolv.conf with nano, but apon reboot it still didplays the old nameserver address

ccrisan, 2016-06-12:
You're not supposed to edit the file manually. How did you set up your network configuration in the first place? Did you configure a static IP? Do you rely on your local DHCP server?

1966tim, 2016-06-12:
I relied on my local DHCP server

ccrisan, 2016-06-13:
Then configure your DHCP server to supply a working DNS server.

1966tim, 2016-06-13:
I have absolutely no idea of how

1966tim, 2016-06-13:
Forget it, I'll just accept that it doesn't work for me and move on. Thanks.

Johnnyb6505, 2016-07-01:
I have the same issue, my DNS is correct, I checked /etc/resolv.conf. I'm also using gmail, I think the issue is you released without adding the necessary packages to send SMTP messages. When I tran and sent a test message from command line I get this: -sh: mail: command not found. Then I tried to install smtp and see this: [root@meye-643fa71a ~]# apt-get install ssmtp
-sh: apt-get: command not found
So no way to fix this? 

Also I'm able to putty to the Pi so I know it's on the network and has connectivity.

[root@meye-643fa71a ~]# ping www.google.com
PING www.google.com (172.217.1.68): 56 data bytes
64 bytes from 172.217.1.68: seq=0 ttl=55 time=23.343 ms
64 bytes from 172.217.1.68: seq=1 ttl=55 time=28.305 ms
64 bytes from 172.217.1.68: seq=2 ttl=55 time=27.079 ms
64 bytes from 172.217.1.68: seq=3 ttl=55 time=28.430 ms
64 bytes from 172.217.1.68: seq=4 ttl=55 time=27.287 ms

Johnnyb6505, 2016-07-01:
I tried to use the mail command because there are about 2,000 Pi pages that have that as a trouble shooting step for mail issues. The ping replies, so now what? Not DNS, is there something else to look at? Config file or some python file? 

 ping smtp.gmail.com
PING smtp.gmail.com (209.85.201.109): 56 data bytes
64 bytes from 209.85.201.109: seq=0 ttl=44 time=40.981 ms
64 bytes from 209.85.201.109: seq=1 ttl=44 time=38.698 ms
64 bytes from 209.85.201.109: seq=2 ttl=44 time=36.994 ms
64 bytes from 209.85.201.109: seq=3 ttl=44 time=39.452 ms
64 bytes from 209.85.201.109: seq=4 ttl=44 time=36.967 ms
64 bytes from 209.85.201.109: seq=5 ttl=44 time=36.707 ms
^C

ccrisan, 2016-07-01:
Make sure you have correctly spelled the mail server name in the settings panel.

Johnnyb6505, 2016-07-01:
I believe this is correct:
smtp.gmail.com
SMTP=587
TLS=yes

I also did not find any messages related to mail to sending in any logs under /var/log is there some other place that I can look?

Also is there some way to send mail from command line to test? Or to bypass the UI?

ccrisan, 2016-07-01:
The settings are correct (assuming that by SMTP=587 you meant PORT). Relevant messages are in/var/log/motion.log. Where did you find the "name or service not known message", by the way?

Unfortunately there's no easy way to manually test the email functionality, but I'll do my best to implement such a button asap.

ccrisan, 2016-07-01:
Ah I thought you were having literally the same problem. In that case, please send me your motion.log and motioneye.log files so I can see if there's anything suspicious in there.

1966tim, 2016-07-01:
I gave up trying to get this to work, didn't realise this thread was still open.Best of luck, it looks great, but I am not meant to use it. Anyway, I am taking a month off from not getting my Raspberry Pis to work, and then I am going to assess whether its worth continuing

1966tim, 2016-07-01:
I gave up on this a month ago. Thanks

Johnnyb6505, 2016-07-02:
Going to reinstall again and see if that helps, I noticed that today as
admin the gmail settings are no longer showing when I browse. Not sure if
this a feature or a bug. Here are my logs. Thanks. The video streaming
works great, if the email part doesn't work then I'll will move on. Might
just put my old OS on the Pi and install all the packages to build up to
where the motionEye is at.

1966tim, 2016-10-30:
I revisited to see if I could resolve the DNS issue.... and its gone. I now get the Email notifications :)

closed 2016-10-30 by 1966tim

jculverwell, 2018-01-09:
I fix for gmail I needed to  :

Log into your google email account and then go to this link: https://www.google.com/settings/security/lesssecureapps and set "Access for less secure apps" to ON.