motioneye-project/motioneyeos #338: Raspberry Pi Zero

Boonheilig, 2016-06-12:
After a clean upload of 20160605 on my zero i get an endless reboot loop (before ui). Up until it searches for wired network. I have a wifi dongle connected to it. Is there a way to skip the hardware check for wired network without ui?

Boonheilig, 2016-06-12:
Or does motioneyeos not support pi zero?

ccrisan, 2016-06-12:
DId you preconfigure a wifi network when you wrote the image to the SD card?

battle79, 2016-06-17:
I am having a similar problem, tried the install file for the pi, and pi2 version installing onto a pi3, setting up wifi, and then  transferring to a pi zero. No luck getting it working on the pi zero. Also tried adding the supplicant file with no luck. Can see supplicant file doing it's thing as I took the SD card out of the pi zero and put in the pi3 and wifi is already configured and connects straight away.

ccrisan, 2016-06-17:
The Zero has nothing to do with PI2 or PI3. Don't try mixing them up (neither OS images, nor the hardware). The Zero is a PI1 A+ in fact so stick with the OS image for the first generation. Place your wpa_supplicant.conf file on the boot partition, connect a monitor to your Zero and tell us what you see.

battle79, 2016-06-17:
ok fresh pi image, i can see the supplicant file isn't working correctly. No network found then it eventually reboots. Supplicant file worked perfectly on my pi3 with the pi3 image.

MrClemGit, 2016-06-20:
I tried to boot a RPi zero with a USB Ethernet hub like this one https://www.adafruit.com/product/2909, but I encountered the same reboot issue. (telling me that the ethernet connection is disconnected.

This USB ethernet hub works well without installation under raspbian jessi. Do I need to add any drivers on buildroot ?

ccrisan, 2016-06-20:
@MrClemGit you cannot and you should not add any drivers to the OS. Can I see your boot.log file?

MrClemGit, 2016-06-20:
I will check that but now I switch to a raspbian jessie to install motioneye.

popeyesfx, 2016-06-23:
i am having the same issue, how do i get the wpa sup file onto the pi, overtime i try using filezilla it doesn't copy, and i am using a model b to try and copy it across.. i don't understand the instructions in the wiki about using settings to set wifi.

ccrisan, 2016-06-23:
Plug the sdcard into your computer and copy your wpa_supplicant.conf file to the boot partition. Then boot it on your Pi Zero. You don't need filezilla at all.

P.S. Make sure to do this operation before the first boot of motionEyeOS.

popeyesfx, 2016-06-23:
Thanks, i worked it out in the end, got a different issue now, have posted that in a separate thread

closed 2016-06-24 by ccrisan

rwkiii, 2017-03-05:
A big clue to most of our problems with the Pi Zero instructions is in one of @ccrisan replies above:

> Plug the sdcard into your computer and copy your wpa_supplicant.conf file to the boot partition. Then boot it on your Pi Zero. You don't need filezilla at all.
> 
> **P.S. Make sure to do this operation before the first boot of motionEyeOS.**

When I read this a bell went off. It seems that @ccrisan is saying that something gets configured on first bootup and if you add the wpa_supplicant.conf AFTER booting without it - it's too late. I've messed with this issue off and on for a week or so. Read everyone else's issues having the same problem. 

Heeding the P.S. in bold above, this worked for me without a hitch. First time.

Write the image to your SD, then add the wpa_supplicant.conf file and then boot your Pi Zero.

ccrisan, 2017-03-05:
@rwkiii that is correct :)

kls629, 2017-08-19:
my noob mistake was naming wpa_supplicant.conf.txt

mensch076, 2017-10-03:
I have a RP Zero and "thePiHut"-dongle connected and saved the wpa_supplicant.conf-file on the sd-card before first booting, but it ended in an endless reboot loop.

"...
 Starting wpa_supplicant: failed (rt2800usb)
 Configuring wireless network: no link
 Configuring wired network: no device
 ethernet: no network connection available, reboot
..."
 
What can I do?
(motioneyeos 20170710)