motioneye-project/motioneyeos #339: Fails to display video or any kind of image

aglproyectos, 2016-06-13:
Hi Calin,

I have tried several times but it keeps not showing a single frame.

I got a Rasperry Pi B with a Raspberry Cam module. I have tried with other tutorials and it works finebut it is becoming a mindbreaking with MotionEyeOS that I find much more interesting to use.

I have seen a previous solved issue with the same content (Fails to display video...) but after careful checking the camera is detected (I deleted it and included once again and became found plus the camera led is correctly managed by the software),

I always find the same message in motioneye.log (ECONNECTREFUSED) and do not find any clue in the rest of log files.
[MotionEye_Fail_Logs.zip](https://github.com/ccrisan/motioneyeos/files/311671/MotionEye_Fail_Logs.zip)

I would appreciate some help...Thanks in advance for it!

ccrisan, 2016-07-03:
Does your camera work with Raspbian (e.g. raspistill/raspivid)?

aglproyectos, 2016-07-06:
Sorry, I found the question just today...

No, they doesn't. I bought a new one just in case it was broken but the result it is exactly the same.
Initially the camera worked fine but maybe has something to do with a firmware update I did while trying different software packages.

Maybe the best way would be to reverse it but I do not know how to do that precisely or, whether it would be better to still use the last firmware available supportting this version of Raspberry Pi.

Thanks for your support..

bortek, 2017-06-01:
I am sure you have solved this issue with your camera hence the silence for almost a year.

closed 2017-06-01 by bortek