motioneye-project/motioneyeos #340: RP2 won't boot after imaging my SD card

anon-e-mouz, 2016-06-13:
My raspberry pi 2 is not booting after using a newly imaged SD card. 
I've used this SD card before with this raspbery pi with other OS's.

It gives a red light, like the boot partition can't be found.

Imaging with Win32DiskImager (started as Administrator) on Windows 7.
Tried 2 SD cards (4gb & 8gb) both with the same issue.

ccrisan, 2016-06-13:
What URL have you used to download the image?

anon-e-mouz, 2016-06-13:
https://github.com/ccrisan/motioneyeos/releases.

Downloaded the latest release 20160605

fettsvett201, 2016-06-13:
Are you getting the "Rainbow Screen"?. I am dealing with the exact same issue. Raspberry pi 2, runs raspbian fine, but cannot boot Motion Eye OS. Tried three different cards. Confirmed camera and Ethernet cable works fine. 

Im using windows 8.1 so no match there.

ccrisan, 2016-06-13:
Guys are you sure you have downloaded the correct image (the one for raspberrypi2) and that you have extracted the downloaded gz file?

fettsvett201, 2016-06-14:
Christ.
I opened up my enclosure to confirm it was a raspberry 2. 

.... Apparently rasp 1 B and rasp 2 B look exactly the same besides a small text difference. Couldn't tell until i opened it up. 

Sorry about not knowing my own equipment. Ill close my issue (335) and apologize for my ignorance.

anon-e-mouz, 2016-06-14:
@ccrisan Yes. Inside the image there's a file called "bcm2709-rpi-2-b.dtb". I extracted the .gz file to an .img file and burned it with the right tool.

Raspbery Pi PCB says "Raspberry Pi (c)2011.12"

@fettsvett201 I have no screen attached so I don't see any rainbow screen.

ccrisan, 2016-06-14:
@anon-e-mouz well if the PCB doesn't mention anything about being a Raspberry PI 2, it's probably a Raspberry PI 1. Try booting the OS image for the first generation of RPi.

anon-e-mouz, 2016-06-14:
@ccrisan When I load the image for the RP1 it boots and it works. Found out I'm using a version 1. That's a bummer. Time to buy a new one! Sorry for the hassle. *facepalm*

processor       : 0
model name      : ARMv6-compatible processor rev 7 (v6l)
BogoMIPS        : 697.95
Features        : half thumb fastmult vfp edsp java tls
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xb76
CPU revision    : 7

Hardware        : BCM2708
Revision        : 000d

ccrisan, 2016-06-14:
Alright, I'll close the issue now.

closed 2016-06-14 by ccrisan