motioneye-project/motioneyeos #342: WiFi EAP (PEAP) settings not being retained on RPi 3 motionEye build 20160605

creepsman, 2016-06-13:
1. wifi is enabled in motionEye GUI
2. /data/etc/wpa_supplicant.conf is update with WPA-EAP settings e.g.
    key_mgmt=WPA-EAP
    ssid="my_SSID"
    pairwise=CCMP
    auth_alg=OPEN
    eap=PEAP
    identity="user"
    password="password"
3. reboot Pi
4. connects to 802.1x network fine - all good at this stage
5. Make any change in the motionEye GUI e.g. a change to the video resolution and the network settings are partially overwritten. The key managment setting is changed back to NONE e.g.:
    key_mgmt=NONE
6. when the Pi is restarted it fails to connect to wifi network - system goes into a booting loop.
7. Connect Pi to wired network and logon. Then manually edit /data/etc/wpa_supplicant.conf and restore the key_mgmt=WPA-EAP setting.
8 restart Pi - all good again until next GUI change and the entry is change back to NONE again.

Question: how do I prevent the GUI from overwritting manual wifi settings?

ccrisan, 2016-06-13:
Unfortunately you can't. I'll need to take care of this and preserve the settings that can be safely kept.

creepsman, 2016-06-13:
Thanks Calin, maybe just have a 'custom' or 'manual' option in the gui dropdown that doesn't modify the .conf???
I guess ideally you could even add the ability to configure the EAP username and password in the GUI, but this would be more complicated as there are few different options e.g. PEAP, EAPTLS etc.

ccrisan, 2016-08-24:
[This commit](https://github.com/ccrisan/motioneye/commit/fdaab12a85238de3644ba3d7751b7759501dee4d) fixes the issue. Basically the key_mgmt field won't be touched at all by the UI unless the psk is completely omitted, in which case, the key_mgmt is set to NONE.

closed 2016-08-24 by ccrisan

rdmarsh, 2018-10-10:
Old thread, but I don't think you need a psk line to use key_mgmt=WPA-EAP. It keeps overwritting it back to none. Until it's overwritten the file works fine.