motioneye-project/motioneyeos #343: Parts of recordings missing.

knobby008, 2016-06-15:
Hello loving the software but I'm running into issues with my stored data. I have my pi set to movie only i have my resolution and frame rate sorted and am happy with the recorded movies. But i am having an issue where if i record 24 hours of movie i only have about 14 hours saved on my hard drive i have a powered 250gb hard drive attached via usb. Over a 24 hour period i am generally getting 2 files both 4.3gb in size. Hope you can help.

ccrisan, 2016-07-03:
What filesystem are you using on your hard drive? You might have hit the 4GB limit of the FAT filesystem.

knobby008, 2016-07-03:
I think that maybe the issue it was formatted as ms dos fat i nave now changed it to ntfs under windows i will run it for a few days fingers crossed. Thanks for taking time to reply.

closed 2017-06-01 by bortek