motioneye-project/motioneyeos #346: stop streaming/recording

jfbdevs, 2016-06-18:
Is there an easy way I could stop streaming/recording without having to shutdown the OS turn off all the cameras?
If it is from the WebUI much better.
I have seen there is an enhancement request for scheduling and it would be great.
But even doing it manually.
Thanks

cuddylier, 2016-06-20:
Maybe the 'Working Schedule' option is what you're looking for if it's the motion detection you want to stop functioning and thus stop recording outside the periods you pick. Not sure if it also works for the constant recording too.

jfbdevs, 2016-06-24:
mmmh I don't need motion detection (monitoring my dogs at the moment lol), and if I disable it also the "Working schedule" gets disabled

cuddylier, 2016-06-24:
Ah...the working schedule must rely on motion then I'd assume.

bortek, 2017-06-01:
Closing his issue due to inactivity. If stil interested please post your question in the forum page.

closed 2017-06-01 by bortek

Shaka-60hp, 2017-12-01:
Hi, start /stop recording could be very usefull
We dont want to constant saving files, just the want to start recording while we are looking at the cameras, and stop when we want too

Is that possible?