motioneye-project/motioneyeos #349: AWB Off gives a green image

AlBaum, 2016-06-20:
Hi,
Completely new here, but discovered motioneyeos and starting to like it !

1. I was wondering whether there is a document somewhere explaining all the setting options. The question mark on the web interface does not work.

2. as the subject says, I am trying to disable the AWB, but when I turn it off I get a green image. Anything I am doing wrong ? I am using a NoIR PiCam on a PI3, with Fast Network Camera enabled.
Thanks
Alain

cuddylier, 2016-06-20:
The question mark beside every option works fine for me on Chrome, you have to hover over it, don't click on it or anything else.

I'll leave 2. for someone else to answer.

AlBaum, 2016-06-21:
Hi cuddylier, thank you for your reply !
 I will check maybe it's due to my browsers configuration.

closed 2017-06-01 by bortek