motioneye-project/motioneyeos #351: Remote motionEye Camera

NizzeHulth, 2016-06-20:
Current Version	20160605

Everything worked great, until I added a camera with the latest version of MotionEye.

When trying to add Remote motionEye Camera, it seems to find the camera, indicating correct camera "name" but results in "authentication failed". ( see picture )

[image]

Any hints?

Björn
Sweden

ccrisan, 2016-06-20:
I don't see any authentication failed message in your screenshot. When do you get that message?

NizzeHulth, 2016-06-20:
It comes after i press OK.
[image]

ccrisan, 2016-06-20:
Can you please send me your motioneye.log file?

NizzeHulth, 2016-06-20:
Here it comes!!

NizzeHulth, 2016-06-20:
[motioneye.pdf](https://github.com/ccrisan/motioneyeos/files/324186/motioneye.pdf)

ccrisan, 2016-06-20:
I have seen many file formats when it came to attaching log files, but this is the first time I'm seeing a PDF :)

Anyways, it does the job. There's indeed a weird authentication problem there. Are you using the latest motionEye 0.32 version?

P.S. I had forgotten to tag the 0.32 on github so you might be running 0.31.5.

NizzeHulth, 2016-06-20:
I couldn't send it as a TXT or LOG, so....

I must take the opportunity to thanks you for a rely good program!

NizzeHulth, 2016-06-20:
I did donwload motioneyeos-raspberrypi2-20160605.img. And the other for Pi 3 of the same date.
Where can I see the version?

NizzeHulth, 2016-06-20:
Its late here. I will be back later.
( Sweden)

ccrisan, 2016-06-20:
It's late here too (Romania) :) I thought you were using vanilla motionEye together with motionEyeOS. So if it's the exact same version running on both units, I'll try to reproduce it myself.

ben-hooper, 2016-06-28:
For what it's worth - I've not been able to get motionEye to recognise any of my 3 Remote motionEye cameras. I use 4 Pi Zeros; one 'hub' and three cameras. The only way I can add the camera-equipped nodes is by calling them Network Camera types during the addition. They're then added as MJPEG video cameras. When I try to add as Remote motionEye cameras, they all fail authentication.

ccrisan, 2016-06-28:
This might be a bug but I haven't had the time to look into it yet. I'll keep you posted.

NizzeHulth, 2016-06-28:
Okey, thanks for trying!!

ccrisan, 2016-07-02:
@ben-hooper can I see a screenshot of the Add Camera dialog with the error when you try to add them as Remote motionEye cameras?

ben-hooper, 2016-07-10:
[image]

ccrisan, 2016-07-10:
@ben-hooper remote motionEye cameras are usually on ports 80 and 8765. Port 8081 is normally used for streaming (i.e. adding it as a network camera).

ben-hooper, 2016-07-10:
@ccrisan - OK, thanks for the input. What's the syntax for declaring the URL? My camera device is called Camera1 and the HTTP Port on that camera-hosting motionEye camera device is configured as port-80 (Expert Settings section), so I'm assuming the URL is http://192.168.0.21:80/cams/Camera1. Is that correct? I'm getting 'Camera Not Found' on that URL. See attached.
[image]
Any further tips please?

ccrisan, 2016-07-10:
http://192.168.0.21 (port 80 is implicit for http).

ben-hooper, 2016-07-10:
I wasn't certain whether you meant to include or exclude the /cams/... part of the URL but ...
[image]
[image]

ccrisan, 2016-07-10:
The second version is the correct one. You must use the root URL, w/o any trailing cams or whatever. The UI will prompt you to select the camera if everything works fine (including the authentication).

Now are you completely sure the password you're using is the admin's password set on your slave motionEyes? Do you run the same version of motionEyeOS on all your 4 units?

ben-hooper, 2016-07-10:
Yes, I'm confident the p/w is being entered correctly - and it is indeed admin's password on the camera unit. I do have the same version of motionEyeOS on all units, yes.

In case it helps, I've also had this variant of response:
[image]
... i.e. where authentication doesn't appear to fail, but no cameras are found. Even though, when I log into the remote camera directly (or when it was set as a Network Camera on the hub unit) - it was clearly streaming OK.

Does this have any dependency on configuration options on the remote camera unit? Should I check any particular settings there?

ccrisan, 2016-07-10:
Have you enabled Fast Network Camera on your slave units?

ben-hooper, 2016-07-10:
Yes, Fast Network Camera was enabled.

ccrisan, 2016-07-10:
Well you can't have FNC active on slaves and add them as Remote motionEye Cameras. The reason is that once you have turned FNC on, the respective device will act as a simple dumb network camera, losing all its "intelligent" abilities provided by the motion daemon (which is no longer used).

If you want to have more FNCs centralized (which you probably do), add them as Network Cameras, using port 8081.

ben-hooper, 2016-07-10:
OK, thanks. 

So FNC features are only accessible when a slave has not been added to a hub (and by inference you're not viewing through a hub but direct on the motionEyeOS device hosting the camera)?

ccrisan, 2016-07-10:
Something like that. Can we consider this issue closed now?

ben-hooper, 2016-07-10:
Yes

Thanks for your support.

ccrisan, 2016-07-10:
With pleasure.

closed 2016-07-10 by ccrisan

soydepr, 2017-03-26:
i having same issue  and not sure why the ip is incorrect 

Mar 26 09:09:25 spycam meyectl[1954]: WARNING: 403 GET /login/?_=1490533768814&_username=_&_signature=538f07a57412bd2050f97402abab0b44d1deadef (192.168.1.22) 1.93ms
Mar 26 09:09:34 spycam meyectl[1954]: ERROR: authentication failed for user pi
Mar 26 09:09:34 spycam meyectl[1954]: WARNING: 403 GET /login/?_=1490533778109&_username=pi&_login=true&_signature=1c2ec39383156e6acdc4768edb0c8c2f254ba776 (192.168.1.22) 4.29ms
Mar 26 09:11:42 spycam rsyslogd-2007: action 'action 17' suspended, next retry is Sun Mar 26 09:12:42 2017 [try http://www.rsyslog.com/e/2007 ]
Mar 26 09:13:18 spycam rsyslogd-2007: action 'action 17' suspended, next retry is Sun Mar 26 09:14:18 2017 [try http://www.rsyslog.com/e/2007 ]
Mar 26 09:13:26 spycam systemd[1]: Starting Session c5 of user pi.
Mar 26 09:13:26 spycam systemd[1]: Started Session c5 of user pi.
Mar 26 09:15:07 spycam meyectl[1954]: WARNING: 403 GET /login/?_=1490534111286&_username=pi&_signature=b7684d6b89460cfe52ca439928f2494dfe1db964 (192.168.1.22) 3.16ms
Mar 26 09:15:07 spycam rsyslogd-2007: action 'action 17' suspended, next retry is Sun Mar 26 09:16:07 2017 [try http://www.rsyslog.com/e/2007 ]
Mar 26 09:15:09 spycam meyectl[1954]: WARNING: 403 GET /login/?_=1490534112871&_username=pi&_signature=47e66db7fae4ea79a4074f6b4dbea809a9233105 (192.168.1.22) 3.28ms
Mar 26 09:15:17 spycam meyectl[1954]: ERROR: authentication failed for user pi
Mar 26 09:15:17 spycam meyectl[1954]: WARNING: 403 GET /login/?_=1490534121089&_username=pi&_login=true&_signature=b93ae70a5a5ba31d9778c174998d88aa4066b2e0 (192.168.1.22) 3.87ms
Mar 26 09:17:01 spycam rsyslogd-2007: action 'action 17' suspended, next retry is Sun Mar 26 09:18:01 2017 [try http://www.rsyslog.com/e/2007 ]
Mar 26 09:17:01 spycam CRON[2081]: (root) CMD (   cd / && run-parts --report /etc/cron.hourly

martinschm, 2019-01-26:
Hi,
I have the same issue. I am using Xiaomi Cams with the dafang hack. They have motion installed. 
How come I can not change the user name from admin to something different ?

I believe it might be different on the Xiaomi hack firmware.