motioneye-project/motioneyeos #358: Pizero and camera

popeyesfx, 2016-06-23:
I have been trying to load os onto a pi zero, it installs fine , but you say in the trouble shooting to use a prepared file to load the wifi automatically. I have tried but it won't let me modify the file.
Im new to raspberry pi so any help would be appreciated.

I am using OS X as my base system to burn and  load files

cheers

popeyesfx, 2016-06-23:
fixed the network issue

pmidhun, 2016-06-24:
@popeyesfx Which OS image did you load into Pi Zero? Did you load motioneyeos-raspberrypi-20160605.img.gz or motioneyeos-raspberrypi3-20160605.img.gz ?

popeyesfx, 2016-06-24:
I used motioneyeos-raspberrypi-20160410.mg as the image and used apple pi baker for OS X to write the card

cheers

ccrisan, 2016-06-29:
I have no experience with OSX but in theory there is no reason why it shouldn't let you edit files on a FAT partition. Have you pulled out and reinserted the SD card after writing the image using apple pi baker? In Linux, it's often necessary to do that in order for the partitions to be re-read.

Tristan79, 2016-06-30:
I have a lot of trouble with OSX Card readers in mac mini and macbooks (read only, corrupted, sectors not written, card will not show up even with brand new cards), also it does not matter if it's a brand new mac mini or an old macbook...  to fix it I use an old external card reader & applepi baker. Reading & Writing works then. This is a known issue back from lion to el capitan. (el capitan is worse, they rewrote the usb drivers such that only 1 in 10 usb hubs work,... *yeah* ;-)

closed 2017-06-01 by bortek