motioneye-project/motioneyeos #359: Crop / Zoom Image Possible?

J-otto, 2016-06-23:
Is it possible to add a setting to crop the image from the camera all of the time?

Personally I would like to set-up RasPi running motioneyeos as a peephole Secuity camera on my front door.  Currently if I put the camera on the peephole >50% of the image is the peephole cylinder and not the image outside the door.  

I could also see image crop being useful for people with privacy concerns who want part of the image always excluded from the image.

ccrisan, 2016-07-03:
I'm afraid this would have to be implemented in motion, rather than in motionEye or motionEyeOS.

closed 2016-07-03 by ccrisan

fernand2000, 2019-04-01:
I was looking for the same thing and found a solution, and as this page is in the top 3 google results I got I'll post my findings here. I have a RPi zero and a RPi camera module v2.1

In the "Video device" tab, in the "Extra Motion Options" window, put
`mmalcam_control_params -roi 0.35,0.25,0.25,0.45`

You'll want to change the numbers according to your needs and this description of the roi (Region of Interest) parameter:

> Allows the specification of the area of the sensor to be used as the source for the preview and capture. This is defined as x,y for the top-left corner, and a width and height, with all values in normalised coordinates (0.0 - 1.0). So, to set a ROI at halfway across and down the sensor, and a width and height of a quarter of the sensor, use: -roi 0.5,0.5,0.25,0.25

muxa, 2020-08-01:
> mmalcam_control_params -roi 0.35,0.25,0.25,0.45

This has no effect in my installation of motioneye on Raspberry Pi (running from docker). Am I missing something?

starbasessd, 2020-08-01:
Try with motioneye on Raspbian or motionEyeOS.
Docker has a lot of issues, as does HomeAssistant.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

desegura, 2021-09-04:
It is not a Docker issue.
ROI is just the definition of the image area where motion needs to find changes. It doesn't really crop and zoom the image.

What we are looking for here is a way to make a zoom in the red frame.

[image]