motioneye-project/motioneyeos #360: Official RPi wifi adapter support

ben-hooper, 2016-06-23:
The official Raspberry Pi wifi adapter appears to be unsupported. I've got two of them and neither appear to be recognised - despite the Pi (using Pi Zero devices) successfully joining the wifi network using Edimax 7811 adapters (I've got 3 of those, all work OK). The official adapter is of course supported in Raspbian and I've used these adapters in Raspbian to validate they're functional.

ccrisan, 2016-06-28:
Can I see your boot.log file?

ccrisan, 2016-06-28:
Also what OS version are you running?

ben-hooper, 2016-06-28:
O/S version is 20160605. I've attached two boot.log files here. One taken after booting with the official RPi adapter (and ethernet, to retrieve the log file) and the other after booting with the Edimax adapter.

[boot(Edimax).log.zip](https://github.com/ccrisan/motioneyeos/files/337042/boot.Edimax.log.zip)
[boot(RPi and ethernet).log.zip](https://github.com/ccrisan/motioneyeos/files/337041/boot.RPi.and.ethernet.log.zip)

ccrisan, 2016-07-01:
The firmware is missing in RPi1 and RPi2 versions. I'll make sure to include it in all variants.

ccrisan, 2016-07-01:
Fixed with acdad088aa22dd9e1e267dfce7e47f217ef2389e.

closed 2016-07-01 by ccrisan