motioneye-project/motioneyeos #361: Stuck with red frame around feed!

kodapaio, 2016-06-24:
[image]

Hi mate!

Dunno what´s the problem is, because I have set the images to be preserved for 1 week only and right now I have 6.3GB of disk usage of 16GB total. But when I check the motioneye.log I found this:

`016-06-19 08:19:07: [motioneye]    ERROR: motion not running, starting it
2016-06-19 08:19:29: [motioneye]    ERROR: motion not running, starting it
2016-06-19 08:21:42: [motioneye]    ERROR: motion not running, starting it
2016-06-19 08:24:24: [motioneye]    ERROR: motion not running, starting it
2016-06-19 08:24:36: [motioneye]    ERROR: motion not running, starting it
2016-06-19 08:24:48: [motioneye]    ERROR: motion not running, starting it
2016-06-19 08:25:50: [motioneye]    ERROR: motion not running, starting it
2016-06-19 08:26:12: [motioneye]    ERROR: motion not running, starting it
2016-06-19 08:26:24: [motioneye]    ERROR: motion not running, starting it
2016-06-19 08:26:46: [motioneye]    ERROR: motion not running, starting it
2016-06-19 08:26:58: [motioneye]    ERROR: motion not running, starting it
2016-06-19 08:27:10: [motioneye]    ERROR: motion not running, starting it
2016-06-19 08:27:22: [motioneye]    ERROR: motion not running, starting it
2016-06-19 08:27:34: [motioneye]    ERROR: motion not running, starting it
2016-06-19 08:27:46: [motioneye]    ERROR: motion not running, starting it
2016-06-19 08:27:58: [motioneye]    ERROR: motion not running, starting it
2016-06-19 08:28:10: [motioneye]    ERROR: motion not running, starting it
2016-06-19 08:28:22: [motioneye]    ERROR: motion not running, starting it
2016-06-19 08:28:34: [motioneye]    ERROR: motion not running, starting it
2016-06-19 08:28:46: [motioneye]    ERROR: motion not running, starting it
2016-06-19 08:28:58: [motioneye]    ERROR: motion not running, starting it
2016-06-19 08:29:10: [motioneye]    ERROR: motion not running, starting it
2016-06-19 08:29:22: [motioneye]    ERROR: motion not running, starting it
2016-06-19 08:29:34: [motioneye]    ERROR: motion not running, starting it
2016-06-19 08:29:46: [motioneye]    ERROR: motion not running, starting it
2016-06-19 08:30:48: [motioneye]    ERROR: motion not running, starting it
2016-06-19 08:31:01: [motioneye]    ERROR: motion not running, starting it
2016-06-19 08:31:13: [motioneye]    ERROR: motion not running, starting it
2016-06-19 08:31:25: [motioneye]    ERROR: motion not running, starting it
2016-06-19 08:31:47: [motioneye]    ERROR: motion not running, starting it
2016-06-19 08:31:59: [motioneye]    ERROR: motion not running, starting it
2016-06-19 08:32:11: [motioneye]    ERROR: motion not running, starting it
2016-06-19 08:32:33: [motioneye]    ERROR: motion not running, starting it
2016-06-19 08:32:55: [motioneye]    ERROR: motion not running, starting it
2016-06-19 08:33:07: [motioneye]    ERROR: motion not running, starting it
2016-06-19 08:33:19: [motioneye]    ERROR: motion not running, starting it
2016-06-19 08:33:31: [motioneye]    ERROR: motion not running, starting it
2016-06-19 08:33:43: [motioneye]    ERROR: motion not running, starting it
2016-06-19 08:33:55: [motioneye]    ERROR: motion not running, starting it
2016-06-19 08:34:07: [motioneye]    ERROR: motion not running, starting it
2016-06-19 08:34:19: [motioneye]    ERROR: motion not running, starting it
2016-06-19 08:34:31: [motioneye]    ERROR: motion not running, starting it`

I´ve found that it takes images, but is not sending them out to me.

Best
Pontus

ccrisan, 2016-06-29:
For some reason the motion daemon stops (every now and then). Common causes are buggy webcam drivers (are you using a USB webcam?) and storage problems (are you using a network share to save media files?).

kodapaio, 2016-08-02:
Update; Na I´m using the RPi camera module v1.0 and saving images to the SD-card.

ccrisan, 2016-08-02:
Check the power adapter. Also make sure you haven't asked too much from your PI. Is this a PI1? You could try to successively disable functionalities such as motion detection, movies or still images and see  if any of those could be the cause of your problem.

kodapaio, 2016-08-02:
This is a RPi 2 model B. But it´s powered down ATM, but has one alike in the basement also that´s works fine.

Will take a look this weekend. :)

alexanderkunin, 2017-04-13:
Having same issue.  Using RPi B. Feed is constant. Power source - 2A. It used to work good with USB, Switched to RPi camera module. At some point this square frame became permanent.
Any findings?

alexanderkunin, 2017-04-13:
Reduced resolution to 640x480 for several minutes, then changed back to 1024x768. Looks like it fixed the problem (at least for now).

ccrisan, 2017-04-14:
Yeap, that can be caused by a poor performance of the board that simply cannot cope with the resolution and framerate set.

closed 2017-04-14 by ccrisan