motioneye-project/motioneyeos #362: change streaming format from Mjpeg to H.264 or Mpeg4

ami2020, 2016-06-24:
Dear 
As you mentioned, motioneyeos uses Mjpeg format for streaming. As you know Mjpeg compress rate is much lower than H.264 format. For example in http://www.supercircuits.com/resources/tools/security-nvr-storage-calculator you could compare 4CIF (704x480) @30fps h.264 for one camera has got 1.54 Mbps but this config with Mjpeg format has got 21.5 Mbps.
The problem is how can I change streaming format from Mjpeg to H.264 or mpeg4?
I think that ffmpeg config must change what is your idea about it?

yours

ccrisan, 2016-10-26:
The streaming format will not change to h264/mp4 any time soon. Sorry about that.

closed 2016-10-26 by ccrisan

tyon2006, 2017-03-14:
I can't seem to find one, but do you have another article where you discuss the complexities of why we cannot make this compression work in this context? This software is excellent, but the bandwidth stops it from being a viable solution for me for this reason.