motioneye-project/motioneyeos #363: Duplicate uploads on Google Drive

jpvillavicencio, 2016-06-24:
Duplicate photos are getting uploaded onto Google Drive.

I've tried rebooting, removing the application key from my google drive.

Tried figuring out why it was happening, but I wasn't able to.

Error log below:

```
2016-06-25 00:36:38: [motioneye]    DEBUG: file "/data/output/garage/2016-06-25/00-34-15.jpg" successfully uploaded
2016-06-25 00:36:38: [motioneye]    DEBUG: loading upload services state from "/data/etc/uploadservices.json"...
2016-06-25 00:36:38: [motioneye]    DEBUG: loaded upload service "gdrive" for camera with id "1"
2016-06-25 00:36:38: [motioneye]    DEBUG: loaded upload service "gdrive" for camera with id "2"
2016-06-25 00:36:38: [motioneye]    DEBUG: uploading file "[/data/output/garage]/2016-06-25/00-34-15.jpg" to gdrive
2016-06-25 00:36:38: [motioneye]    DEBUG: size of "/data/output/garage/2016-06-25/00-34-15.jpg" is 0.1MB
2016-06-25 00:36:38: [motioneye]    DEBUG: mime type of "/data/output/garage/2016-06-25/00-34-15.jpg" is "image/jpeg"
2016-06-25 00:36:38: [motioneye]    DEBUG: requesting https://www.googleapis.com/upload/drive/v2/files?uploadType=multipart
2016-06-25 00:36:39: [motioneye]    DEBUG: 200 HEAD / (127.0.0.1) 1.64ms
2016-06-25 00:36:39: [motioneye]    DEBUG: file "/data/output/garage/2016-06-25/00-34-15.jpg" successfully uploaded
2016-06-25 00:36:39: [motioneye]    DEBUG: loading upload services state from "/data/etc/uploadservices.json"...
2016-06-25 00:36:39: [motioneye]    DEBUG: loaded upload service "gdrive" for camera with id "1"
2016-06-25 00:36:39: [motioneye]    DEBUG: loaded upload service "gdrive" for camera with id "2"
2016-06-25 00:36:39: [motioneye]    DEBUG: uploading file "[/data/output/garage]/2016-06-25/00-34-16.jpg" to gdrive
2016-06-25 00:36:39: [motioneye]    DEBUG: size of "/data/output/garage/2016-06-25/00-34-16.jpg" is 0.1MB
2016-06-25 00:36:39: [motioneye]    DEBUG: mime type of "/data/output/garage/2016-06-25/00-34-16.jpg" is "image/jpeg"
2016-06-25 00:36:39: [motioneye]    DEBUG: requesting https://www.googleapis.com/upload/drive/v2/files?uploadType=multipart
2016-06-25 00:36:41: [motioneye]    DEBUG: file "/data/output/garage/2016-06-25/00-34-15.jpg" successfully uploaded
2016-06-25 00:36:41: [motioneye]    DEBUG: loading upload services state from "/data/etc/uploadservices.json"...
2016-06-25 00:36:41: [motioneye]    DEBUG: loaded upload service "gdrive" for camera with id "1"
2016-06-25 00:36:41: [motioneye]    DEBUG: loaded upload service "gdrive" for camera with id "2"
2016-06-25 00:36:41: [motioneye]    DEBUG: uploading file "[/data/output/garage]/2016-06-25/00-34-16.jpg" to gdrive
2016-06-25 00:36:41: [motioneye]    DEBUG: size of "/data/output/garage/2016-06-25/00-34-16.jpg" is 0.1MB
2016-06-25 00:36:41: [motioneye]    DEBUG: mime type of "/data/output/garage/2016-06-25/00-34-16.jpg" is "image/jpeg"
2016-06-25 00:36:41: [motioneye]    DEBUG: requesting https://www.googleapis.com/upload/drive/v2/files?uploadType=multipart
```

ccrisan, 2016-06-29:
It's not a duplicate. It's a new file having the same name as the previous one. The reason comes from the fact that the naming pattern includes time variables up to seconds. If your system takes more than one photo within a second, they will all share the same name. If you must have multiple photos taken within the same second, use %q in their names.

closed 2016-06-29 by ccrisan

jpvillavicencio, 2016-07-03:
I have set the system to only take a photo once per second but there are still multiple photos being uploaded to google drive. Usually in sets of 4-6. I understand if only 2 photos were being uploaded but 3-4 is causing an issue. 

As shown below, the photos have the same timestamp and trigger comments. When uploaded to Google, they have the same name and size. So it is the same photo being uploaded multiple times.

If you need assistance in finding the error, PM me and I'll be happy to help.

[image]

ccrisan, 2016-07-04:
Indeed, if you have, say, 4 files sharing the same name, what will actually happen is that the file with that name will be scheduled for 4 updates a bit later (usually after 5 seconds). So whatever the content of the file is after 5 seconds, that's going to be uploaded 4 times. The only workaround for now is to include %q (frame number) in the file name so that each picture will have a distinct name.

PeterPedro, 2019-12-06:
I have an issue because each captured frame (still images) gets a lot of repeated pictures, so I end up like with 7 thousand images in my Drive account, and the internal storage folder gets the right amount of pictures (like 650).

Windows or Google Drive rename the files like (1) (2) and so on, but the images are exactly the same, some files get more than 20 copies some get 3 or 4, seems very random.

PeterPedro, 2019-12-16:
I was having this issue and then added the %q, the images are not repeated now but the problem is that too many images are being stored in Google Drive, my internal flash memory gets the right amount of pictures but as soon as I enable Google Drive, the camera starts adding more images than it should, it adds multiple images per second to GD but in my configuration it shouldn't, in theory it should add the same amount of images it adds to my flash drive.

lucianaohira, 2020-12-02:
My issue is... I am using %v at the filename... And It seems that after setting GoogleDrive to save all the files, all of them are being uploaded with the same name 01.jpg. At Google Drive I have many 01.jpg, and at MotionEyeOS I´ve got only one 01.jpg... Am I missing some configuration?

starbasessd, 2020-12-02:
I can take a look at it tomorrow. 
I need your filename string, Which version of motionEyeOS.

lucianaohira, 2020-12-02:
Hey, hi! Thanks in advance!!!
My filename string is just %v
motionEye Version 0.42.1
Motion Version 4.2.2+gitUNKNOWN OS
Version motionEyeOS 20200606

starbasessd, 2020-12-02:
Manual snapshots are always 01.jpg
motion detected snapshots will increment, but on reboot the number resets.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

lucianaohira, 2020-12-03:
Ok, I hadn´t realize about the manual/motion detection snapshot. Now I am using %v with motion detection, but I am really having many image copies uploaded at my GoogleDrive, the maximum number I counted was something like 90 copies of the same file... It must be some crazy config I did!