motioneye-project/motioneyeos #364: raspi camera doesn't show up in local cameras

michaelarz, 2016-06-26:
Hi, just installed motioneyeos-raspberrypi2-20160605.img on rpi2, works pretty well with usb cam but not with raspberry camera. 
[image]
Tried 2 different cameras (both pinoir) and two different raspberry pi2 boards. Before this cameras and boards was working with another os. All logs may be found attached. 

If needed can try with rpi1 or another version tomorrow.

[logs.zip](https://github.com/ccrisan/motioneyeos/files/333846/logs.zip)

Appreciate any help!

ccrisan, 2016-06-29:
Things don't look too good in your log files. For starters I recommend using a different port than 80 when forwarding connections from the internet. I can see many requests to `/common/info.cgi` which indicate standard HTTP attacks. I'm not saying this is the root cause of your problem :) Judging by other errors I have seen in your logs, at this point I recommend rewriting the OS image from scratch.

michaelarz, 2016-06-30:
Hi! Thanks a lot, probably was just a bug during initial flashing or I was so stupid to connect csi camera to wrong port - who knows now. After flashing again it works. And it was no attack - I would never make http(w/o s) service accessible from internet. Just me connecting too often from my laptop:)

closed 2016-06-30 by michaelarz