motioneye-project/motioneyeos #368: RTSP 401 erroneous response

tvb, 2016-06-28:
I receive the following error:
[image]

while I am sure the details are correct since I can connect to the cam via RTSP with VLC. 

I am running MotionEyeOs 20160605 on a raspberrypi2

tvb, 2016-06-29:
Does anyone has an idea? Running MJPEG isnt the prettiest.

ghostyroasty, 2016-06-30:
I have the same issue using an Amcrest prohd outdoor wifi bulletcam.

ccrisan, 2016-07-01:
There have been a few users complaining about authentication issues with RTSP. However I have never been able to reproduce the problem. Would any of you be so kind and temporarily forward a port to such a RTSP camera with authentication enabled, so I can run some tests and see how credentials are passed?

ghostyroasty, 2016-07-01:
E-mail sent to your gmail address.

ccrisan, 2016-07-02:
Fixed with https://github.com/ccrisan/motioneye/commit/c3f900a1412698568c13f81cee3ed24a18070e19.

closed 2016-07-02 by ccrisan