motioneye-project/motioneyeos #376: build failure, cannot stat... no such file

skippyV, 2016-07-04:
Hello, I tried to build the latest sources (commit 306b2ec) with Kubuntu, and undoubtedly did something wrong because I had to keep adding execute privilege to many sh scripts during the process. But thought I was almost there when I saw this error. It's as if the build was trying to install it on the system - I think.
Still a Linux novice.
I followed the github instructions which basically said "./build.sh raspberrypi2".
Any advice?
[image]
thanks

ccrisan, 2016-07-04:
That's odd, considering [this](https://github.com/ccrisan/motioneyeos/blob/master/board/common/postscript.sh#L10). Can you run ./build.sh raspberrypi2 in a clean shell and send me the entire output?

skippyV, 2016-07-04:
./build.sh raspberrypi2 2>&1 | tee raspBldOut.txt  from a new console. Is
that what you meant by "clean" console?

[7m>>>   Finalizing target directory[27m
rm -rf /home/skippy/raspberryPi_related/motioneyeos/output/raspberrypi2/target/usr/lib/luarocks
find /home/skippy/raspberryPi_related/motioneyeos/output/raspberrypi2/target/usr/lib/python2.7 -name '*.py' -print0 | xargs -0 rm -f
mkdir -p /home/skippy/raspberryPi_related/motioneyeos/output/raspberrypi2/target/etc/network/
( echo "# interface file auto-generated by buildroot"; echo ; echo "auto lo"; echo "iface lo inet loopback"; ) > /home/skippy/raspberryPi_related/motioneyeos/output/raspberrypi2/target/etc/network/interfaces
rm -rf /home/skippy/raspberryPi_related/motioneyeos/output/raspberrypi2/target/usr/include /home/skippy/raspberryPi_related/motioneyeos/output/raspberrypi2/target/usr/share/aclocal \
		/home/skippy/raspberryPi_related/motioneyeos/output/raspberrypi2/target/usr/lib/pkgconfig /home/skippy/raspberryPi_related/motioneyeos/output/raspberrypi2/target/usr/share/pkgconfig \
		/home/skippy/raspberryPi_related/motioneyeos/output/raspberrypi2/target/usr/lib/cmake /home/skippy/raspberryPi_related/motioneyeos/output/raspberrypi2/target/usr/share/cmake
find /home/skippy/raspberryPi_related/motioneyeos/output/raspberrypi2/target/usr/{lib,share}/ -name '*.cmake' -print0 | xargs -0 rm -f
find /home/skippy/raspberryPi_related/motioneyeos/output/raspberrypi2/target/lib /home/skippy/raspberryPi_related/motioneyeos/output/raspberrypi2/target/usr/lib /home/skippy/raspberryPi_related/motioneyeos/output/raspberrypi2/target/usr/libexec \
		\( -name '*.a' -o -name '*.la' \) -print0 | xargs -0 rm -f
rm -rf /home/skippy/raspberryPi_related/motioneyeos/output/raspberrypi2/target/usr/share/gdb
rm -rf /home/skippy/raspberryPi_related/motioneyeos/output/raspberrypi2/target/usr/share/zsh
rm -rf /home/skippy/raspberryPi_related/motioneyeos/output/raspberrypi2/target/usr/man /home/skippy/raspberryPi_related/motioneyeos/output/raspberrypi2/target/usr/share/man
rm -rf /home/skippy/raspberryPi_related/motioneyeos/output/raspberrypi2/target/usr/info /home/skippy/raspberryPi_related/motioneyeos/output/raspberrypi2/target/usr/share/info
rm -rf /home/skippy/raspberryPi_related/motioneyeos/output/raspberrypi2/target/usr/doc /home/skippy/raspberryPi_related/motioneyeos/output/raspberrypi2/target/usr/share/doc
rm -rf /home/skippy/raspberryPi_related/motioneyeos/output/raspberrypi2/target/usr/share/gtk-doc
rmdir /home/skippy/raspberryPi_related/motioneyeos/output/raspberrypi2/target/usr/share 2>/dev/null
make[1]: [target-finalize] Error 1 (ignored)
find /home/skippy/raspberryPi_related/motioneyeos/output/raspberrypi2/target -type f \( -perm /111 -o -name '*.so*' \) -not \( -name 'libpthread*.so*' -o -name 'ld-*.so*' -o -name '*.ko' \) -print0 | xargs -0 /home/skippy/raspberryPi_related/motioneyeos/output/raspberrypi2/host/usr/bin/arm-buildroot-linux-gnueabihf-strip --remove-section=.comment --remove-section=.note 2>/dev/null || true
if test -d /home/skippy/raspberryPi_related/motioneyeos/output/raspberrypi2/target/lib/modules; then \
		find /home/skippy/raspberryPi_related/motioneyeos/output/raspberrypi2/target/lib/modules -type f -name '*.ko' -print0 | \
		xargs -0 -r /home/skippy/raspberryPi_related/motioneyeos/output/raspberrypi2/host/usr/bin/arm-buildroot-linux-gnueabihf-strip --remove-section=.comment --remove-section=.note --strip-unneeded; fi
find /home/skippy/raspberryPi_related/motioneyeos/output/raspberrypi2/target/lib -type f -name 'libpthread*.so*' | \
		xargs -r /home/skippy/raspberryPi_related/motioneyeos/output/raspberrypi2/host/usr/bin/arm-buildroot-linux-gnueabihf-strip --remove-section=.comment --remove-section=.note --strip-debug
find /home/skippy/raspberryPi_related/motioneyeos/output/raspberrypi2/target/lib -type f -name 'ld-*.so*' | \
		xargs -r /home/skippy/raspberryPi_related/motioneyeos/output/raspberrypi2/host/usr/bin/arm-buildroot-linux-gnueabihf-strip --remove-section=.comment --remove-section=.note --strip-debug
test -f /home/skippy/raspberryPi_related/motioneyeos/output/raspberrypi2/target/etc/ld.so.conf && \
		{ echo "ERROR: we shouldn't have a /etc/ld.so.conf file"; exit 1; } || true
test -d /home/skippy/raspberryPi_related/motioneyeos/output/raspberrypi2/target/etc/ld.so.conf.d && \
		{ echo "ERROR: we shouldn't have a /etc/ld.so.conf.d directory"; exit 1; } || true
mkdir -p /home/skippy/raspberryPi_related/motioneyeos/output/raspberrypi2/target/etc
( \
		echo "NAME=Buildroot"; \
		echo "VERSION=2016.02-g306b2ec-dirty"; \
		echo "ID=buildroot"; \
		echo "VERSION_ID=2016.02"; \
		echo "PRETTY_NAME=\"Buildroot 2016.02\"" \
	) >  /home/skippy/raspberryPi_related/motioneyeos/output/raspberrypi2/target/etc/os-release
[7m>>>   Copying overlay board/common/overlay[27m
[7m>>>   Copying overlay board/raspberrypi2/overlay[27m
[7m>>>   Executing post-build script board/common/postscript.sh[27m
board/common/postscript.sh: 31: board/common/postscript.sh: board/common/cleanups.sh: Permission denied
[7m>>>   Generating root filesystem image rootfs.tar[27m
rm -f /home/skippy/raspberryPi_related/motioneyeos/output/raspberrypi2/build/_fakeroot.fs
rm -f /home/skippy/raspberryPi_related/motioneyeos/output/raspberrypi2/target/THIS_IS_NOT_YOUR_ROOT_FILESYSTEM
rm -f /home/skippy/raspberryPi_related/motioneyeos/output/raspberrypi2/build/_users_table.txt
[... cut ...]
/usr/bin/install -m 0644 support/misc/target-dir-warning.txt /home/skippy/raspberryPi_related/motioneyeos/output/raspberrypi2/target/THIS_IS_NOT_YOUR_ROOT_FILESYSTEM
:: creating boot loop device
20+0 records in
20+0 records out
20971520 bytes (21 MB) copied, 0.029557 s, 710 MB/s
:: creating boot filesystem
mkfs.fat 3.0.26 (2014-03-07)
Loop device does not match a floppy size, using default hd params
:: mounting boot loop device
:: copying boot filesystem contents
cp: cannot stat ‘/home/skippy/raspberryPi_related/motioneyeos/board/raspberrypi2/../../output/raspberrypi2/images//boot/*’: No such file or directory

ccrisan, 2016-07-04:
Yeap, something like that. For some reason, your `board/common/postscript.sh` is not executable:

    board/common/postscript.sh: 31: board/common/postscript.sh: board/common/cleanups.sh: Permission denied

Without that your build will never succeed. We need to identify what's going on with all those missing executable flags in your scripts.

skippyV, 2016-07-05:
Thanks for pointing that out. I wasn't being observant...
So I added execute privilege to the sh scripts in board/common,
board/raspberrypi2 and writeimage.sh
then I got an error: "realpath: command not found"
So after googling I did "sudo apt-get install coreutils", which did
nothing. Was already installed.
Then I tried "sudo apt-get install realpath", which did install
'realpath_1.19_amd64.deb'. Even though this suggestion on a post was given
a down-vote. (I tried it anyways. Shooting in the dark)
Then my last result is this. Which did result in an image being built (size
220 MiB), but there were still a few errors generated in the output.
Is this is a successful build?

[7m>>>   Finalizing target directory[27m
rm -rf /home/skippy/raspberryPi_related/motioneyeos/output/raspberrypi2/target/usr/lib/luarocks
find /home/skippy/raspberryPi_related/motioneyeos/output/raspberrypi2/target/usr/lib/python2.7 -name '*.py' -print0 | xargs -0 rm -f
mkdir -p /home/skippy/raspberryPi_related/motioneyeos/output/raspberrypi2/target/etc/network/
( echo "# interface file auto-generated by buildroot"; echo ; echo "auto lo"; echo "iface lo inet loopback"; ) > /home/skippy/raspberryPi_related/motioneyeos/output/raspberrypi2/target/etc/network/interfaces
rm -rf /home/skippy/raspberryPi_related/motioneyeos/output/raspberrypi2/target/usr/include /home/skippy/raspberryPi_related/motioneyeos/output/raspberrypi2/target/usr/share/aclocal \
		/home/skippy/raspberryPi_related/motioneyeos/output/raspberrypi2/target/usr/lib/pkgconfig /home/skippy/raspberryPi_related/motioneyeos/output/raspberrypi2/target/usr/share/pkgconfig \
		/home/skippy/raspberryPi_related/motioneyeos/output/raspberrypi2/target/usr/lib/cmake /home/skippy/raspberryPi_related/motioneyeos/output/raspberrypi2/target/usr/share/cmake
find /home/skippy/raspberryPi_related/motioneyeos/output/raspberrypi2/target/usr/{lib,share}/ -name '*.cmake' -print0 | xargs -0 rm -f
find /home/skippy/raspberryPi_related/motioneyeos/output/raspberrypi2/target/lib /home/skippy/raspberryPi_related/motioneyeos/output/raspberrypi2/target/usr/lib /home/skippy/raspberryPi_related/motioneyeos/output/raspberrypi2/target/usr/libexec \
		\( -name '*.a' -o -name '*.la' \) -print0 | xargs -0 rm -f
rm -rf /home/skippy/raspberryPi_related/motioneyeos/output/raspberrypi2/target/usr/share/gdb
rm -rf /home/skippy/raspberryPi_related/motioneyeos/output/raspberrypi2/target/usr/share/zsh
rm -rf /home/skippy/raspberryPi_related/motioneyeos/output/raspberrypi2/target/usr/man /home/skippy/raspberryPi_related/motioneyeos/output/raspberrypi2/target/usr/share/man
rm -rf /home/skippy/raspberryPi_related/motioneyeos/output/raspberrypi2/target/usr/info /home/skippy/raspberryPi_related/motioneyeos/output/raspberrypi2/target/usr/share/info
rm -rf /home/skippy/raspberryPi_related/motioneyeos/output/raspberrypi2/target/usr/doc /home/skippy/raspberryPi_related/motioneyeos/output/raspberrypi2/target/usr/share/doc
rm -rf /home/skippy/raspberryPi_related/motioneyeos/output/raspberrypi2/target/usr/share/gtk-doc
rmdir /home/skippy/raspberryPi_related/motioneyeos/output/raspberrypi2/target/usr/share 2>/dev/null
make[1]: [target-finalize] Error 1 (ignored)
find /home/skippy/raspberryPi_related/motioneyeos/output/raspberrypi2/target -type f \( -perm /111 -o -name '*.so*' \) -not \( -name 'libpthread*.so*' -o -name 'ld-*.so*' -o -name '*.ko' \) -print0 | xargs -0 /home/skippy/raspberryPi_related/motioneyeos/output/raspberrypi2/host/usr/bin/arm-buildroot-linux-gnueabihf-strip --remove-section=.comment --remove-section=.note 2>/dev/null || true
if test -d /home/skippy/raspberryPi_related/motioneyeos/output/raspberrypi2/target/lib/modules; then \
		find /home/skippy/raspberryPi_related/motioneyeos/output/raspberrypi2/target/lib/modules -type f -name '*.ko' -print0 | \
		xargs -0 -r /home/skippy/raspberryPi_related/motioneyeos/output/raspberrypi2/host/usr/bin/arm-buildroot-linux-gnueabihf-strip --remove-section=.comment --remove-section=.note --strip-unneeded; fi
find /home/skippy/raspberryPi_related/motioneyeos/output/raspberrypi2/target/lib -type f -name 'libpthread*.so*' | \
		xargs -r /home/skippy/raspberryPi_related/motioneyeos/output/raspberrypi2/host/usr/bin/arm-buildroot-linux-gnueabihf-strip --remove-section=.comment --remove-section=.note --strip-debug
find /home/skippy/raspberryPi_related/motioneyeos/output/raspberrypi2/target/lib -type f -name 'ld-*.so*' | \
		xargs -r /home/skippy/raspberryPi_related/motioneyeos/output/raspberrypi2/host/usr/bin/arm-buildroot-linux-gnueabihf-strip --remove-section=.comment --remove-section=.note --strip-debug
test -f /home/skippy/raspberryPi_related/motioneyeos/output/raspberrypi2/target/etc/ld.so.conf && \
		{ echo "ERROR: we shouldn't have a /etc/ld.so.conf file"; exit 1; } || true
test -d /home/skippy/raspberryPi_related/motioneyeos/output/raspberrypi2/target/etc/ld.so.conf.d && \
		{ echo "ERROR: we shouldn't have a /etc/ld.so.conf.d directory"; exit 1; } || true
mkdir -p /home/skippy/raspberryPi_related/motioneyeos/output/raspberrypi2/target/etc
( \
		echo "NAME=Buildroot"; \
		echo "VERSION=2016.02-g306b2ec-dirty"; \
		echo "ID=buildroot"; \
		echo "VERSION_ID=2016.02"; \
		echo "PRETTY_NAME=\"Buildroot 2016.02\"" \
	) >  /home/skippy/raspberryPi_related/motioneyeos/output/raspberrypi2/target/etc/os-release
[7m>>>   Copying overlay board/common/overlay[27m
[7m>>>   Copying overlay board/raspberrypi2/overlay[27m
[... cut ...]
:: reading partition offsets
:: destroying disk loop device
:: creating boot loop device
:: copying boot image
40960+0 records in
40960+0 records out
20971520 bytes (21 MB) copied, 0.275083 s, 76.2 MB/s
:: destroying boot loop device
:: creating root loop device
:: copying root image
368640+0 records in
368640+0 records out
188743680 bytes (189 MB) copied, 2.17272 s, 86.9 MB/s
:: destroying root loop device
:: /home/skippy/raspberryPi_related/motioneyeos/output/raspberrypi2/images/motioneyeos-raspberrypi2.img is ready

ccrisan, 2016-07-05:
You tell me if it's a successful build :) Those few errors are normal. The size of 220MB is correct. If you gzip it you should get something around 65MB.

skippyV, 2016-07-06:
Unfortunately it was not successful. The raspberrypi logo is displayed on
the top but the output is just "can't open /dev/ttylogin: No such file or
directory".
So I just went ahead and used your latest pre-release (20160705).
So far so good.
Thanks for all your help!

ccrisan, 2016-07-06:
Can you please send me your boot.log file from your build? I'd like to debug and fix these issues, as people should be able to build their own OS images without too much trouble.

skippyV, 2016-07-06:
I have one final dumb question. I got a shared Windows folder successfully
accessed by motioneyeos using File Storage-> Network Share configuration.
(That was a big win for me - yeah!). But I don't understand how to make it
autosave clips. For example if I have motion triggered on and want to save
a pic every 2 seconds, how is the resulting files saved to my network
share? My current config has no files generated.

I'll have to work on that boot.log request when I'm back home. On travel
for a few days. So may be a week or more before I'm at that computer with
time to spare.

ccrisan, 2016-07-06:
You can't have pictures saved both every two seconds and triggered by motion. You'll need to enable Still Images and choose between the two modes.

skippyV, 2016-07-06:
Ok. But since I didn't have Still Images enabled, and did motion detection
enabled, shouldn't I have seen video files being generated in my Network
Share?

ccrisan, 2016-07-06:
Videos (aka movies) are controlled by the Movies section while pictures (aka still images) are controlled by the Still Images section. They are completely separated and can be enabled individually. Whether they are stored on the local disk or remotely on a network share is controlled by the options in the File Storage section.

Is it a bit clearer now?

skippyV, 2016-07-13:
Yes. I understand. Thank yo for your patience.
I believe now that my Pi2 is a bit unstable.  Over the past 3 days I've
gotten multiple video files uploaded to Google Drive except for the 2nd day
when no files were uploaded. And since I know my neighbor walked on the
patio to check on things that particular day - there was definitely motion.
And on the days that videos are captured it appears that many videos are
"missing". For example I might be unloading the car for several minutes but
only the first 2 mins are captured. My maximum movie length I set to 2
minutes... but if motion is continuous in the field of view, shouldn't the
camera just keep making 2 min videos, one after another?
Which is why I believe it is a hardware problem. Maybe I'll get a Pi3 and
start over.

closed 2017-06-01 by bortek