motioneye-project/motioneyeos #378: Unable to Obtain Dropbox Key

leeford, 2016-07-04:
Hi,

I've got the latest build running on a Pi 2. All working well, except when I select "Upload Media Files", select "Dropbox", click "Obtain Key" I get an error on the newly opened page:

{"error": "unknown upload service dropbox"}

Let me know if you need anything further.

Thanks.

leeford, 2016-07-04:
Just to add, with "Google Drive" I get:

{"error": "unknown upload service gdrive"}

Thanks.

ccrisan, 2016-07-04:
Is this 20160703?

jeancazes, 2016-07-04:
I have the same problem 
same message 
v : 20160703

try with chrome and firefox.(just in case)

ccrisan, 2016-07-04:
It's a known bug. I have already fixed it upstream with https://github.com/ccrisan/motioneye/commit/5def02069cc8c8818fc6e322561e542e592724a4. Thanks for reporting it though.

closed 2016-07-04 by ccrisan

leeford, 2016-07-04:
OK thank you.