motioneye-project/motioneyeos #379: No video on fresh install (RPi 3)

RickyYoder, 2016-07-05:
Just wrote the OS onto the card and booted it for the first time. Waited for it to do all of its prepwork. When I go to view the cam from the IP, the preview/stream is black. The camera is the Raspberry Pi 2 camera module w/IR.

RickyYoder, 2016-07-05:
Logitech USB webcam is working though. Not sure why the Pi camera isn't.

RickyYoder, 2016-07-05:
After trying to apply new settings to the camera, it appears to work upon refreshing, but the image quickly fades to black.

ccrisan, 2016-07-05:
Is this an original RPi camera module? Do you have the automatic brightness option enabled?

RickyYoder, 2016-07-05:
Yes, it's an original model. Also, when I turned automatic brightness on and set the FPS higher, the image would fade into white, fade into black, and fade back into white and then fade back into black in a loop. It's very strange.

closed 2017-06-01 by bortek