motioneye-project/motioneyeos #380: Camera live images not saved at external file storage

Tsubakura, 2016-07-05:
After installing MotionEye from scratch, I've manage to set up most of the things to make it work.
The problem though, is the live feed of the camera doesn't get saved at the File Storage I've specified on the device.

The external file storage is a Win10 shared folder, which I've mounted on the raspberry through fstab.
The shared folder is correctly recognized from the device, live streaming also seems to work.
The main problem is that I can't get my motioneye to save the live stream to my external storage or actually anywhere.

It seems to have worked before, so I don't know why it suddenly stopped working out of the blue.

ccrisan, 2016-07-05:
Please attach your config files (motion.conf, thread-1.conf) *after removing sensitive data*.

Tsubakura, 2016-07-05:
Thanks for the quick reply.

I have both the files here, the sensitive data has been replaced with asterisks.

motion.conf:
 # @enabled on
 # @show_advanced on
 # @normal_password *****
 # @admin_username *****
 # @admin_password *****
 # @normal_username *****

thread thread-1.conf
control_html_output on
setup_mode on
control_localhost on
control_port 7999

thread-1.conf:
 # @webcam_resolution 100
 # @upload_subfolders on
 # @upload_server 
 # @enabled on
 # @network_server 192.168.0.70
 # @upload_username 
 # @motion_detection off
 # @upload_port 
 # @upload_location 
 # @preserve_movies 7
 # @network_username *****
 # @id 1
 # @upload_password 
 # @upload_method post
 # @working_schedule_type outside
 # @webcam_server_resize on
 # @network_password *****
 # @upload_service dropbox
 # @name Camera1
 # @preserve_pictures 30
 # @storage_device custom-path
 # @network_share_name CameraRecords
 # @upload_enabled off
 # @working_schedule 

webcam_motion on
height 768
threshold 0
quality 100
noise_level 31
pre_capture 2
noise_tune off
hue 0
saturation 0
ffmpeg_variable_bitrate 0
ffmpeg_video_codec msmpeg4
text_changes on
movie_filename %Y-%m-%d/%H-%M-%S
auto_brightness off
rotate 0
brightness 0
lightswitch 0
framerate 30
snapshot_filename 
snapshot_interval 0
webcam_maxrate 30
target_dir /mnt/Storage
text_double on
webcam_quality 100
post_capture 4
on_picture_save /usr/bin/python /usr/local/lib/python2.7/dist-packages/motioneye/meyectl.pyc relayevent -c /etc/motioneye/motioneye.conf picture_save %t %f
on_movie_end /usr/bin/python /usr/local/lib/python2.7/dist-packages/motioneye/meyectl.pyc relayevent -c /etc/motioneye/motioneye.conf movie_end %t %f
text_left Camera1
webcam_port 8081
locate_motion_style redbox
contrast 0
locate on
videodevice /dev/video0
output_all on
ffmpeg_cap_new off
gap 10
on_event_end /usr/bin/python /usr/local/lib/python2.7/dist-packages/motioneye/meyectl.pyc relayevent -c /etc/motioneye/motioneye.conf stop %t
output_normal on
text_right %Y-%m-%d\n%T
on_event_start /usr/bin/python /usr/local/lib/python2.7/dist-packages/motioneye/meyectl.pyc relayevent -c /etc/motioneye/motioneye.conf start %t
ffmpeg_bps 7499999
jpeg_filename %Y-%m-%d/%H-%M-%S
minimum_motion_frames 10
webcam_localhost off
width 1024
max_mpeg_time 0

ccrisan, 2016-07-06:
1. Your config file seems to belong to an old version of motion. Is this really motionEyeOS or is it a motionEye installation + an official 3.2.12 motion?

2. You have `ffmpeg_cap_new off` which means that you have movies turned off in the settings panel.

Tsubakura, 2016-07-06:
This device has Raspbian with motionEye installed + an official motion.
Is there a difference between that and motionEyeOS?

ccrisan, 2016-07-06:
Well there are many differences. motionEyeOS is ready to be used but does not allow advanced customizations available with a regular Raspbian. Since you opened the issue in motionEyeOS's issue tracker, I just assumed you were actually using motionEyeOS.

You never said anything about number 2.

Tsubakura, 2016-07-08:
Ah, thats good to know.
I actually had issues running motionEyeOS because it doesn't seem to be able to detect the camera.

Anyway, what is the best course of action for me?
Should I just reinstall the device?

ccrisan, 2016-07-08:
You still didn't mention anything about number 2. I'll repeat the question for you: You have `ffmpeg_cap_new off` which means that you have movies turned off in the settings panel. With movies turned off, you won't... well.. have movies.

Tsubakura, 2016-07-08:
I changed the settings now:

 # @webcam_resolution 100  
 # @upload_subfolders on
 # @upload_server 
 # @enabled on
 # @network_server 192.168.0.70
 # @upload_username 
 # @motion_detection off
 # @upload_port 
 # @upload_location 
 # @preserve_movies 7
 # @network_username *****
 # @id 1
 # @upload_password 
 # @upload_method post
 # @working_schedule_type outside
 # @webcam_server_resize on
 # @network_password *****
 # @upload_service dropbox
 # @name Camera1
 # @preserve_pictures 30
 # @storage_device custom-path
 # @network_share_name CameraRecords
 # @upload_enabled off
 # @working_schedule 

webcam_motion on
height 768
threshold 0
quality 100
noise_level 31
pre_capture 2
noise_tune off
hue 0
saturation 0
ffmpeg_variable_bitrate 0
ffmpeg_video_codec msmpeg4
text_changes on
movie_filename %Y-%m-%d/%H-%M-%S
auto_brightness off
rotate 0
brightness 0
lightswitch 0
framerate 30
snapshot_filename 
snapshot_interval 0
webcam_maxrate 30
target_dir /mnt/Storage
text_double on
webcam_quality 100
post_capture 4
on_picture_save /usr/bin/python /usr/local/lib/python2.7/dist-packages/motioneye/meyectl.pyc relayevent -c /etc/motioneye/motioneye.conf picture_save %t %f
on_movie_end /usr/bin/python /usr/local/lib/python2.7/dist-packages/motioneye/meyectl.pyc relayevent -c /etc/motioneye/motioneye.conf movie_end %t %f
text_left Camera1
webcam_port 8081
locate_motion_style redbox
contrast 0
locate on
videodevice /dev/video0
output_all on
ffmpeg_cap_new on
gap 10
on_event_end /usr/bin/python /usr/local/lib/python2.7/dist-packages/motioneye/meyectl.pyc relayevent -c /etc/motioneye/motioneye.conf stop %t
output_normal on
text_right %Y-%m-%d\n%T
on_event_start /usr/bin/python /usr/local/lib/python2.7/dist-packages/motioneye/meyectl.pyc relayevent -c /etc/motioneye/motioneye.conf start %t
ffmpeg_bps 7499999
jpeg_filename %Y-%m-%d/%H-%M-%S
minimum_motion_frames 10
webcam_localhost off
width 1024
max_mpeg_time 0

The device is a Raspbian with probably motionEye installation + an official motion. The version on the settings show 0.30rc1. There is only 1 camera installed on this device, there is no number 2.

ccrisan, 2016-07-08:
Number 2 was the second item in my comment a few days ago.

Tsubakura, 2016-07-08:
ffmpeg_cap_new should be on now.
It doesn't change anything though...

ccrisan, 2016-07-08:
Have you tried to disable the shared folder and use the local disk as storage (in File Storage)? The first step is to get your movies to be created locally.

Tsubakura, 2016-09-10:
Sorry for the late answer, I was very busy.
I have redirected the file storage to a local folder in /mnt/Test and I can confirm that the pi doesn't create the movies at all.

ccrisan, 2016-09-11:
@Tsubakura please attach your config files (again). But this time please *attach* them rather than *pasting* them. Also, make sure to remove sensitive data from them.

closed 2017-06-01 by bortek