motioneye-project/motioneyeos #381: Use motionEyeOS  behind proxy

nextgen1979, 2016-07-07:
Hi,  it's possible to set proxy setting in motioneyeos ? 

Thanks in advance 

Nicolas

ccrisan, 2016-07-07:
Nope. There is no proxy support in motionEyeOS and there are no plans to add one.

dazzag24, 2016-07-07:
+1 for http_proxy support.

shanester64, 2016-07-11:
Why does base path setting exist (although it does not work)?

[image]

ccrisan, 2016-07-11:
@shanester64 what do you mean it doesn't work? Have you tried it? By the way, it's meant for *reverse proxy* usage and has nothing to do with internal motionEye clients using a normal HTTP proxy (which is indeed not supported).

shanester64, 2016-07-11:
I understand it is for reverse proxy. I have tried it on both stable and current pre-release on rp2.For example, upon changing the base path to /cams and rebooting I would get an error message not found (http://IP:port/cams).  Thanks,Ken Shane

ccrisan, 2016-07-11:
You actually need to use a reverse proxy. Did you use Apache or Nginx or anything of that sort? Of course nothing will work anymore if you're just appending /cams to your motionEyeOS machine URL. motionEyeOS will still be accessed internally by your reverse proxy with its root URL / but all the static resources (images, scripts and so on) will be expected to be found at e.g. /cams.

I have been using it this way for a year now so I know for a fact that it works.

shanester64, 2016-07-12:
Thanks for your input. I do use nginx, however I never configured it due to the errors (via internal). I will configure and test. Being somewhat new to nginx, based upon your input, can you access both internally and externally (nginx)?

Thank you,
Ken Shane

ccrisan, 2016-07-12:
Nope, unless you use no base URL.

shanester64, 2016-07-12:
So your confirming that changing the base url/path does not work.  Thanks,Ken Shane

ccrisan, 2016-07-12:
I do not confirm that. I believe you have misunderstood how it works and how reverse proxies work in general.

MiGeppert, 2017-02-04:
Hi ccrisan

Would you share your nginx configuration file part concerning the motioneyeos-proxy (and "special settings" in motioneye if necessary)?

I tried to proxy https://"myservername"/webcam on nginx to http://"webcamip"/ but I thinks I'm too stupid for this. I spent hours searching the internet without success!
The "best" thing I got is a white background with parts of the left side menu and no chance to login (instead of the dark background with login window). When I enter "/webcam" to the "Base Path" in the expert settings, I get the "white background" thing even when entering http://"webcamip"/ and have to log in via ssh to correct/delete the "Base Path" line in /data/etc/motioneye.conf to get http access to my camera.

Can you help me please?

Many thanks for making that great motioneye(os) software.

Best regards
MiGeppert

bortek, 2017-06-01:
Closing his issue due to inactivity. If stil interested please post your question in the forum page.

closed 2017-06-01 by bortek

hiphophurray, 2018-12-28:
Hi, 

I've just tried to configure MotionEyeOS (Odroid XU4) and NGINX/Let'sEncrypt/DynamicDNS (Raspberry Pi 3B+) within my environment, internally (http://10.1.1.2:80) I can access the MEOS system perfectly and view my cameras however when I go to my external domain (https://sub.mydomain.com) address over port 443 (I do have 443>80 redirect on via CertBot/NGINX) I cannot see the MEOS web page correctly. I have looked and looked online for an answer for this but cannot find the answer. The last question above from MiGeppert seems to be quite close to what i'm looking to do. 

Can anyone please assist? I'm not willing to open port 80 to my server VLAN from the outside world hence the reasoning for NGINX.

I must also mention i want to access the MEOS web page from the base URL (/).

[image]

Thanks for any answers that may come.
Frank

ccrisan, 2019-01-02:
@hiphophurray can we see your nginx config (at least the relevant part)?

shanester64, 2019-01-03:
> Hi all, hi ccrisan
> I'm sad that I got no answer to my question(s) I asked almost two years ago.
> But I still want to use motioneye behind my Nginx-proxy (to get rid of my "direct port forwarding" to my motioneye raspberry).
>
> Attached you find my trial-and-error part of the nginx config.
> My final goal is to reach http-motioneye via "https-subdirectory" (https://"myserveraddress.com"/xwebcam/ -> http://192.168.2.150)
>
> 	location /xwebcam/ {
>                         # rewrite ^/xwebcam(/.*)$ $1 break;
> 			proxy_pass http://192.168.2.150:80/;
> 			proxy_read_timeout 120s;
> 			access_log off;
> 			proxy_set_header Host $host;
> 			proxy_set_header Host $host:$server_port;
> 			proxy_set_header X-Real-IP $remote_addr;
> 			proxy_set_header X-Forwarded-For $proxy_add_x_forwarded_for;
> 			proxy_set_header X-Forwarded-Proto https;
> 		}
> Would be nice if you can help me (and Frank/hiphophurray) by sharing a working nginx configuration.
>
> Thanks and best regards
> MiGeppert

Here is my partial config..replace IP with your IP address 

location /cams/ {
        proxy_pass http://IP:8765/;
        proxy_read_timeout 120s;
        #access_log off;
        proxy_set_header Host $host;
        proxy_set_header X-Forwarded-Host $server_name;
        proxy_set_header X-Real-IP $remote_addr;
        proxy_set_header X-Forwarded-For $proxy_add_x_forwarded_for;
        # SSL proxying headers
        proxy_set_header    X-Forwarded-Proto   $scheme;
        proxy_set_header    X-Forwarded-Ssl     on;        }

        location /mjpeg {
        proxy_pass http://IP:5888/;
        proxy_read_timeout 86400;
        #access_log off;
        proxy_set_header Host $host;  
        proxy_set_header X-Forwarded-Host $server_name;  
        proxy_set_header X-Real-IP $remote_addr;  
        proxy_set_header X-Forwarded-For $proxy_add_x_forwarded_for; 
        # SSL proxying headers
        proxy_set_header    X-Forwarded-Proto   $scheme; 
        proxy_set_header    X-Forwarded-Ssl     on; 
        }

Thank you,
Ken Shane

MiGeppert, 2019-01-03:
Hi Ken Shane
Thanks for your config file - this works as long as I use the unsecure http:// connection (http://"myserveraddress.com"/xwebcam/), but when I use the secure connection (https://"myserveraddress.com"/xwebcam/) I got only a white page with four lines:
1. an empty pull down menu,
2. "Apply" (no link)
3. "webcam" (I think the name of my Motioneye-Cam)
4 "motionEye" (link to https://"myserveraddress.com"/xwebcam/)

I don't want to use my user:password with http :-(.

Note: both config files for Nginx were modified ("default" and "ssl-default")

Best regards
MiGeppert

shanester64, 2019-01-03:
Here is the 1st part of my default config (replace mydomain.com with yours).

# Redirect http to https.    server {        listen 80;        server_name mydomain.com;        return 301 https://$host$request_uri;    }
# HTTPS Server    server {        listen 443 ssl;        server_name mydomain.com;
#        root /var/www/mydomain.com/html;        root /usr/share/nginx/html;        index index.html index.htm;
        ssl on;        ssl_certificate /etc/letsencrypt/live/mydomain.com/fullchain.pem;        ssl_certificate_key /etc/letsencrypt/live/mydomain.com/privkey.pem;
        ssl_session_timeout 1d;        ssl_session_cache shared:SSL:50m;        ssl_stapling on;        ssl_stapling_verify on;        add_header Strict-Transport-Security max-age=15768000;
        ssl_protocols TLSv1 TLSv1.1 TLSv1.2;        ssl_ciphers 'ECDHE-RSA-AES128-GCM-SHA256:ECDHE-ECDSA-AES128-GCM-SHA256:ECDHE-RSA-AES256-GCM-SHA384:ECDHE-ECDSA-AES256-GCM-SHA384:DHE-RSA-AES128-GCM-SHA256:DHE-DSS-AES128-GCM-SHA256:kEDH+AESGCM:ECDHE-RSA-AES128-SHA256:ECDHE-ECDSA-AES128-SHA256:ECDHE-RSA-AES128-SHA:ECDHE-ECDSA-AES128-SHA:ECDHE-RSA-AES256-SHA384:ECDHE-ECDSA-AES256-SHA384:ECDHE-RSA-AES256-SHA:ECDHE-ECDSA-AES256-SHA:DHE-RSA-AES128-SHA256:DHE-RSA-AES128-SHA:DHE-DSS-AES128-SHA256:DHE-RSA-AES256-SHA256:DHE-DSS-AES256-SHA:DHE-RSA-AES256-SHA:AES128-GCM-SHA256:AES256-GCM-SHA384:AES128-SHA256:AES256-SHA256:AES128-SHA:AES256-SHA:AES:CAMELLIA:DES-CBC3-SHA:!aNULL:!eNULL:!EXPORT:!DES:!RC4:!MD5:!PSK:!aECDH:!EDH-DSS-DES-CBC3-SHA:!EDH-RSA-DES-CBC3-SHA:!KRB5-DES-CBC3-SHA';        ssl_prefer_server_ciphers on;        ssl_dhparam /etc/ssl/certs/dhparam.pem;
        location ~ /.well-known {                allow all;       }
#       redirect server error pages to the static page /50x.html        error_page 500 502 503 504 /50x.html;        location = /50x.html {               root /var/www/mydomain.com/html;      }

 

Thanks,

Ken Shane

 
 

   
   - an empty pull down menu,
   - "Apply" (no link)