motioneye-project/motioneyeos #383: Wifi Dongle Not Detected

adriankwil, 2016-07-08:
So i just got a TP-Link WN823N dongle to add to my Rpi2, and motionEyeOS is not recognising it. Is this an issue with missing drivers? or am i missing some obvious step.
I've tried booting into a clean install both with and without the dongle attached, rebooting with it attached after setting up the wireless settings in the web ui and all that shows under ifconfig is ethernet. Would someone be able to help with this?

Thanks

I'm using version 20160705 (Pre-Release)

ccrisan, 2016-07-08:
1) Does it work with Raspbian?
2) Can you attach your boot.log file?

adriankwil, 2016-07-08:
Here is the boot log, and i'll install raspbian and get back to you. (Damn that was a quick reply!)
[boot.log.txt](https://github.com/ccrisan/motioneyeos/files/354241/boot.log.txt)

ccrisan, 2016-07-08:
Yeap, the line `Configuring wireless network: no device` indicates motionEyeOS does not recognize the device. Now let's see if Raspbian does.

adriankwil, 2016-07-08:
So I installed, updated and upgraded raspbian but the dongle is not detected. I tested that it does actually work on my pc but it also didn't work until I installed the drivers.

adriankwil, 2016-07-08:
This is very strange as on amazon it said that it is compatible with the pi. And online everywhere I've seen it simply says that this adapter is compatible out of the box with the pi.

closed 2017-06-01 by bortek

daniele777, 2017-09-10:
I have the same issues motioneyes search only cable connection don't try to configure wi fi dongle... how i can fix? unfortunaly i have only wi fi at home..