motioneye-project/motioneyeos #385: Problems with videostream (ghosting?)

Goudtik, 2016-07-09:
Has anyone got any idea how to solve these issues with this networkcamera? It is used wireless and eventually there are also green horizontal stripes in the video (not only on this cam by the way but this one keeps giving me problems) I hope this link works since that is so much easier than trying to disc ribs it with words: [IMG]http://i68.tinypic.com/1676psi.png[/IMG]

Goudtik, 2016-07-09:
the green stripes I was talking about: [IMG]http://i66.tinypic.com/1zr27h2.png[/IMG] I really hope someone can give me some pointers..... Thanks!

micky2171, 2016-07-09:
This looks like a webcam issue, do you have any other cameras to try?

ghostyroasty, 2016-07-13:
I have the same problem. I'm using an Amcrest ip2m-842w. Using Blue Iris and viewing the live feed in other software does not have these issues.

Goudtik, 2016-07-13:
@micky i have 4 ip cameras running but this one is the only one that has this problems. @ghostyroasty same here, glad that I am not the only one. Hopefully we can solve this....

ostermayer, 2016-08-13:
yes also experiencing this on ics realtime and ubiquiti ip wired cameras

closed 2017-06-01 by bortek