motioneye-project/motioneyeos #386: Camera Not Compatible?

RickyYoder, 2016-07-09:
Just bought a Waveshare camera with IR lights. When I try to add it from the Web UI the dropdown list says "no cameras!". It plugs in just like the regular Pi camera. And there don't seem to be any special drivers for it. 

But if I have to return this one that's okay. What are some wide lens infrared cameras that will work out of the box with motioneyeos?

ccrisan, 2016-07-09:
Does it work with Raspbian?

RickyYoder, 2016-07-09:
I did not test in Raspbian yet. But I can.

RickyYoder, 2016-07-09:
I have established a proper connection in MotionEyeOS.

I'm getting an error though. 

"error 0 waiting for frame completion". The camera is lit, but it isn't recording.

RickyYoder, 2016-07-09:
I'm trying to get the camera to work in Raspbian at the moment. I wrote a basic Python script after installing python3-picamera:

```
import picamera

cam = picamera.PiCamera()

cam.start_preview()
```

The only thing that happens when I run the module is that IDLE says:

"=================RESTART================="

I've enabled the camera through raspi-config and I've upgraded all packages. All of which were part of the instructions to set up this camera. Could it be that I have a bad camera?

ccrisan, 2016-07-10:
1. When I said trying it in Raspbian I meant using raspistill or raspivid commands to see if the camera actually works. I never thought you would write some code to that end :)

2. You can safely ignore the `error 0 waiting for frame completion` message.

closed 2017-06-01 by bortek