motioneye-project/motioneyeos #387: Timelapse stuck at 0%?

micky2171, 2016-07-09:
[image]

Is there any way that I can view and cancel other timelapses too?

ccrisan, 2016-07-10:
Unfortunately not.

micky2171, 2016-07-10:
Now it says 'Failed to create Timelapse' after a reboot?

ccrisan, 2016-07-10:
Without starting the timelapse process? Just out of nowhere?

micky2171, 2016-07-10:
I restarted the Pi and clicked Timelapse, 30 seconds, 30FPS, OK
It now gets to 1% and then errors. The same thing happens when trying to create a ZIP of every image

ccrisan, 2016-07-10:
Most likely you have too many pictures to be processed and both the zip and the timelapse procedure fails for two possible reasons: (1) the process takes too much time or (2) the free memory is not available. You could attach your motioneye.log here so that we clear this up. However the solution in your case is to work with less pictures or to choose a more powerful hardware.

closed 2017-06-01 by bortek