motioneye-project/motioneyeos #389: Banana PI camera

simbiotica-it, 2016-07-10:
hi, thanks for the great job.
I bought a bananaPi but is not recognized , is supported ?
[http://www.tme.eu/it/Document/187887186b98a8f78b47da2774a34f4c/BPI-CAMERA.pdf](url)
ciao!

ccrisan, 2016-07-10:
No idea. Can you log in through SSH and see if you have the `/dev/video0` device file present?

simbiotica-it, 2016-07-10:
the device /dev/video0 isn't present

simbiotica-it, 2016-07-10:
[http://wiki.lemaker.org/BananaPro/Pi:Camera_Module](url)
I tried to load the modules on this page but the system responds : not found in modules.dep

ccrisan, 2016-07-10:
Does it work with any of the official operating systems?

simbiotica-it, 2016-07-10:
Tested with Lubuntu distribution, the WebCam works fine.
installed with:
sudo modprobe ov5640
sudo modprobe sun4i_csi0
tested with GUVCview

ccrisan, 2016-07-11:
Unfortunately it seems that mainline integration of the BPi CSI camera is still a work in progress: http://linux-sunxi.org/Mainlining_Effort#Major_drivers.

ektus, 2016-08-21:
I've got a Banana Pi and CSI camera as of this weekend, and ran into the same problem. The motioneyeos image (version 20160705) lacks the ov5640 and sund4i_sci0 driver modules. With Bananian as OS (latest version, Jessie-based), the camera works (sort of), and motion produces images when started stand-alone without further configuration, but MotionEye doesn't find the internal camera due to errors. I'll report them over on the MotionEye tracker.
Would it be worthwhile copying over the two modules from the Bananian install? If so, how?

closed 2017-06-01 by bortek