motioneye-project/motioneyeos #392: Issue when updating Pi's

mehtaman, 2016-07-11:
Hi all,

I have some trouble updating my Raspberry Pi's and was wondering if anyone else has had an issue?

I have a Pi 2 & Pi 3 board.
Both were running stable build 20160605

When I updated to Pre-Release 20160703 both failed to update correctly and I had to reformat the SD card and setup MotionEyeOS from scratch.

The same issue occurred on my Pi 3 when updating from 20160703 to 20160705.
I have not tried to update my Pi 2 yet.

Did anyone end up having to format / reconfigure their systems when updating?

ccrisan, 2016-07-11:
Is the issue reproducible? If yes, can you send me your log files after a failed upgrade?

mehtaman, 2016-07-11:
Hi ccrisan,

I'm a bit stuck trying to get the log files from my Pi 3.

I can ping the device but the web browser shows "{"error": "internal server error"}" - I have tried restarting the system several times.

Putty says the connection failed.
I use a windows system and Samba also failed to connect \\{ip address}\

I will try later today to reformat the SD card with 20160703 and update to 20160703 through the GUI.
Hopefully I can replicate/catch the log files for you.

ccrisan, 2016-07-11:
Please do. Or otherwise plug in your SD card to a PC running Linux and copy the entire /var/log directory to your PC.

mehtaman, 2016-07-11:
Hi ccrisan, 

I think I have attached my log files...

[Downloads.zip](https://github.com/ccrisan/motioneyeos/files/357890/Downloads.zip)

ccrisan, 2016-07-11:
Yes you have. However the funny thing is that there's no trace of actual upgrading of your system. The following lines indicate that you have clicked on the Update button once:

    2016-07-08 11:09:59: [motioneye]    DEBUG: fetching https://api.github.com/repos/ccrisan/motioneyeos/releases?_=1467972599...
    2016-07-08 11:10:01: [motioneye]    DEBUG: available versions: 20160705, 20160703, 20160605, 20160410, 20160320, 20160306, 20160221, 20160109, 20151230, 20151206, 20151129, 20151103, 20151025, 20151021, 20151011

Have you then actually proceeded with the update?

When it comes to the internal server error issue, there are two errors that pop out:

    2016-07-11 18:51:37: [motioneye]    ERROR: failed to open media preview image file: cannot identify image file <StringIO.StringIO instance at 0x742c0300>

and

    2016-07-11 18:50:55: [motioneye]    ERROR: failed to get config for remote camera 1 on http://10.44.1.68:9000/config/1: request timed out

For the first one, well... make sure you have enough space on your SD card (is it full, btw?). Other than that, hell knows. Somehow the movie preview image which was supposed to be a jpeg, well.. isn't a jpeg.

For the second one, the remote motionEye camera that you have presumably configured, does it work as expected? Does it respond to requests?

mehtaman, 2016-07-11:
Hi ccrisan, 

Thank you for checking the files...

My only other guess is that the WiFi was not strong enough when I tried to update the system.
The Pi 3 on a wired connection updated fine
The Pi 2 is working again after a power cycle

The SD cards have lots of space, one is 32GB and another is 8GB... both systems are set to delete the files after 1 day so it shouldn't fill up.

After updating the Pi 3 and restarting the Pi 2 both respond on the web interface.

Things seem ok now :)

ccrisan, 2016-07-12:
I guess we'll never know :)

closed 2016-07-12 by ccrisan