motioneye-project/motioneyeos #394: Heartbeat - watchdog timer

veames, 2016-07-11:
Hi Caylen,

Was hoping you could tell me how to enable/add the watchdog timer feature as was in motionpi?

I am looking for the device to reboot every time it is unable to connect with the backend router/file repository. This fixes the situation where the pi looses wifi and has to be powered off manually.

thanks

ccrisan, 2016-07-11:
It's enabled by default.

TheKitro, 2016-07-28:
Hi 
When I stop the Server with:
meyectl stopserver -b -c /etc/motioneye.conf

after 120 sek I got an reboot.
How can i deactivate the autoreboot, when motioneye is manually forced offline.

thanks

veames, 2016-07-28:
Why not use systemd? #systemctl stop motioneye

TheKitro, 2016-07-29:
I got following results:

[root@meye-40 startstop]# systemctl stop motioneye.service
-sh: systemctl: command not found
[root@meye-40 startstop]# sh systemctl stop motioneye.service
sh: systemctl: No such file or directory
[root@meye-40 startstop]# systemctl stop motioneye
-sh: systemctl: command not found
[root@meye-40 startstop]# sh systemctl stop motioneye
sh: systemctl: No such file or directory
[root@meye-40 startstop]#

seems not working.

ccrisan, 2016-07-29:
If you're on motionEyeOS you don't have systemctl, and you probably shouldn't stop motionEye or any other services manually :) However, if you must do that, add `meyewatch_disable="true"` to your `/data/etc/watch.conf`.

closed 2017-06-01 by bortek