motioneye-project/motioneyeos #395: Raspberry Pi Zero rebooting

Kriss1988, 2016-07-12:
Hi
I was using motionEyeOS on RPi3 for some time and it worked fine. Recently I got my RPi zero and wanted to test it. I've downloaded motioneyeos-raspberrypi-20160705.img.gz, wrote it on sd and started.
I'm using micro usb -> rj45 adapter: http://goo.gl/2J7ErC and CSI camera.
Rasp keeps rebooting every few seconds.
Logs:
[logs.zip](https://github.com/ccrisan/motioneyeos/files/360040/logs.zip)

Did anyone had similar problem?

ccrisan, 2016-07-12:
Your messages file says, among other stuff, the following:

    user.notice ethernet: disconnected for 20 seconds, rebooting

That basically means your ethernet connection, once established, some how broke.

Kriss1988, 2016-07-12:
Yes, but i have no idea why. I've tried to set up static ip, but didn't help. I've tested raspbian and it works fine, no connection lose.

ccrisan, 2016-07-12:
Hmm the adapter probably doesn't behave as motionEyeOS expects. Could you please run Raspbian on your Zero and temporarily give me remote SSH access so I can run some tests?

Ikeoth, 2016-08-04:
I also had a RPI Zero that constantly rebooted stating the Ethernet was disconnected. The initial run of the OS was when there was no USB peripherals attached including the USB Ethernet. After formatting and reapplying the OS image to the SD card and having the UISB Ethernet adapter attached at initial run it ran the initial build process normally and started the OS and Ethernet perfectly fine.

Great OS! Many Thanks!

stefanides, 2016-08-19:
Hi, I am noticing the same behaviour, RPI B works ok, the same uSD card in RPI zero gives the same log message 
`user.notice ethernet: disconnected for 20 seconds, rebooting`
The interesting thing is, that the message is read from the log via that ethernet = it is still alive. 

I have two different usb ethernet modules, the one works ok without any problem
```
[    2.727497] usb 1-1.3: New USB device found, idVendor=0b95, idProduct=772b
[    6.786461] asix 1-1.3:1.0 eth0: register 'asix' at usb-20980000.usb-1.3, ASIX AX88772B USB 2.0 Ethernet, 00:00:00:00:57:XX
[    6.788098] usbcore: registered new interface driver asix
```
just the other one make troubles: 
```
[  414.024644] usb 1-1.4: New USB device found, idVendor=0fe6, idProduct=9700
[  414.083633] dm9601 1-1.4:1.0 eth1: register 'dm9601' at usb-20980000.usb-1.4, Davicom DM96xx USB 10/100 Ethernet, 00:e0:4c:53:44:XX
[  414.094307] usbcore: registered new interface driver dm9601
[  414.114661] usbcore: registered new interface driver sr9700
```
tested on the latest motioneye (20160705)

ccrisan, 2016-08-20:
Try the 20160801 prerelease and see if it works.

stefanides, 2016-08-21:
Thanks, the prerelease works ok on Pi Zero. (Pi B looks that it cannot handle extra USB ethernet card on eth1 = it neither does not boot with it nor it can survive switching of default route to it later. Maybe some /etc/network/interfaces config helps). 
Anyway I suppose the reboot timeout is configurable in UI = and the default 20 secs could be "dangerous" in installations where the ethernet connection is not secured. Any ethernet disconnection (cable, switch, router..) will then cause camera to be all the time almost down (cca 20 secs monitoring + cca 1 min reboot). At least the configuration tooltip could warn about that :-)

By the way, there looks to be also some issue with self updating on PI zero - it works for me only when running on PI B. But that is another topic :-)

closed 2017-06-01 by bortek