motioneye-project/motioneyeos #398: Camera runs fine for a few days then it stop attaching the pictures?

ts4s, 2016-07-14:
I've been running this for some time now and I have noticed that after some time has passed, i'll get email without the photos attached. what would cause this?  At first I thought it would be the 1 gig usb drive that i'm using to save the photo to so i change it to a 32 gig and i am still have the same issue? Attached is the log file after i updated it today, need to due one more update(see other post).

[motioneye.txt](https://github.com/ccrisan/motioneyeos/files/362882/motioneye.txt)

ccrisan, 2016-07-14:
Usually when there are too many pictures, parsing them takes too long and motionEye decides to skip adding pictures, sending the email right away.

ccrisan, 2016-07-15:
Nope. You'll need to enable the Preserve options (for Still Images, Movies or both, depending on what you're using).

ts4s, 2016-07-16:
Already have that setting. Noticed today I received emails but not photos after preforming updates 2 days ago. Didn't change any setting at that time.

Today I saw where I could update to the lastest, and am reloading the config file. Ran the test button on the email, and that worked! So we'll see what this weekend brings?

ccrisan, 2016-07-16:
@ts4s if you see error messages like `[sendmail] ERROR: timeout waiting for the media listing process to finish` in your motion.log, you probably have too many pictures, even with the preserve option.

ts4s, 2016-07-16:
So does that mean I need to clear out the stored photos?

ccrisan, 2016-07-16:
It means that you should configure your system so that it has to deal with less photos at once. I have no idea to what value you have set your preserve option, but you can reduce it to one day, if you haven't done it already.

If you're using motion detection and you have configured your still images to be saved based on motion detection, you should also check about how many motion events you have a day. Reducing the motion detection threshold will also reduce the number of false motion events and will obviously reduce the number of pictures taken due to motion events.

Given that during motion, all possible frames are saved, it could be an idea to reduce the capture framerate of your video device.

As a last resort, you could switch to better hardware (a better board, a faster SD card).

ts4s, 2016-07-18:
since reload the config, everything seem to work. will check it again on friday cause it takes about a week for it to do it thing (and usually starts acting up when i'm gone, heading out for a few days next sunday).

 i have started to play with the motion threshold but it's slow progress... Noticed that the camera (noir v1) is not picking me up once i exit the vehicle and walk towards it. it pick up the vehicle once it's almost to the house and park, not when it pulls into the driveway. however it will pick up a car sitting way of in the distance with it's head light pointing at it in the dark. so it may be time to switch over to the v2 of the pi cam.

bortek, 2017-08-16:
@ccrisan Let me reopen this for a moment. When it scans for a pictures does it goe through the whole /data/output/Camera1  and all subdirectories ? Righ now I have ~5000 pictures spread in 20 directories. But in todays directory it's 1113 images which I think is not much at all. Is the scanning done on the actual day directory or the whole /data/output/Camera1 including subdirs? And what does the scanning do, is it looking for a file in the time span where the motion was triggered ( I am guessing) ?

ccrisan, 2017-08-17:
It does actually scan through the entire /data/output/Camera1 directory. There's no other way (that I can think of) to narrow the scanning in the case of an event.

bortek, 2017-08-17:
Wouldn't it be much faster to scan in the current days directory? They have the standard naming 2017-01-01 or whatever the fortmat is. So if the code can search in that directory tree only it would be much faster, wouldn't it?

I have gone down to 3 days files bus it still it times out. Email notification without images are almost useless IMO.

ccrisan, 2017-08-17:
That's the problem: there's no standard naming. Everyone is free to choose their own name pattern. I agree that part of code should be improved. I just don't know how.

bortek, 2017-08-17:
But these are saved in the variables called 

movie_filename %Y-%m-%d/%H-%M-%S
picture_filename %Y-%m-%d/%H-%M-%S

So by reading these variables the code can list the format of the directory name. And the media_path holds the first portion of the path  (/data/output)

ccrisan, 2017-08-17:
We could optimize the search for when patterns are left as default (the format that begins with %Y-%m-%d). And in the rest of the cases we could fallback to the current full-search method.

bortek, 2017-08-17:
I think that will work for most of the cases. How do we know the default format is used. We could have a dropdown (like Preserve Pictures dropdown) with "default format" and "custom" choices and when you select custom then the text field pops up under the drop down with default and editable format.

ccrisan, 2017-08-17:
We simply compare with "%Y-%m-%d/%H-%M-%S" :)

bortek, 2017-08-17:
Whenever you have time to implement this I can do test a patch and test on my system. This is if you want to do a test before you include it in the next release.

ccrisan, 2017-08-26:
https://github.com/ccrisan/motioneye/commit/b4e05e03150b9bdae95ede45f6ba11a459507b8e adds this optimization.

closed 2017-08-26 by ccrisan

Instagit, 2018-08-23:
This is still a problem if you want to keep more than a few days of pictures. Motion detection seems to work better with higher FPS (it lowers the detection errors from subtle lighting changes etc.) which adds to the problem.

Just for clarification - is this the function that is causing the problems?

https://github.com/ccrisan/motioneye/blob/master/motioneye/mediafiles.py#L366

If it is, I'll try to optimize it when I got some time.

erik80, 2020-01-04:
I wrote this workaround script:
```
[root@pi1 ~]# cat /data/output/mv_cam.sh
#!/bin/bash
target_dir=$(grep target_dir /data/etc/camera-*.conf | sed -e 's/[^ ]* //' -e 's;/$;;' | head -n 1);
bak_dir="${target_dir}_bak" 
mkdir -p "$bak_dir"
echo "$(date): $target_dir -> $bak_dir"
ls -pt1 "${target_dir}" | grep "/$" | awk '{if(NR>1) print}' | xargs -r -t -I{} mv -v "${target_dir}/{}" "${bak_dir}/{}"

```
which moves all directories except the newest one to backup dir (e.g.: /home/ftp/sdcard/Camera1/ -> /home/ftp/sdcard/Camera1_bak/ ). I run it from crontab every hour.
`[root@pi1 ~]# crontab -l
0 * * * * /data/output/mv_cam.sh >> /data/output/mv_cam.log
`
Probably it will not work with multiple cameras in one rpi. And it will not work if you have lot of files in the newest dir.