motioneye-project/motioneyeos #400: Initial motion detection

dexe3, 2016-07-18:
When the device is restarted or settings are adjusted the (v4l2) camera is restarted, after which it detects motion. I think it's an issue caused by initial exposure adjustments or similar, which causes the device to detect motion. Problem is that when email or images saving is enabled, the device freezes and doesn't recover until I ramp up the motion detection treshhold. With images saving and email notifications turned off, the device doesn't freeze and it becomes clear that it detects motion for ~3-5 seconds after which it decreases and motion is detected as intended. Sadly I can not find a way to detect actual motion without detecting motion during the start. I've already set/tried brightness/contrast/saturation and noise detection on manual values.

Basically my request is to add a feature which delays the motion tracking after starting the device or after applying settings.

stllunatic, 2016-07-18:
As a possible work around you could use the working hours option to schedule out when you want motion detection to begin. Once you can verify that the environment is stable you can apply different settings as needed.

dexe3, 2016-07-18:
I thought about doing that as well. I'm just afraid that when the device restarts or crashes for whatever reason my inbox will get spammed with emails of motion detection and no way for me to easily fix it, due to not being at home. Also, after applying those schedule changes the camera still restarts so the problem would remain. I could only set it to be off when I start it for the first time and then leave it like that. It would be so ideal if I could just let it ignore motion for ~10 seconds, every time the camera is started :)

closed 2017-06-01 by bortek