motioneye-project/motioneyeos #402: Ability to add SSL/TLS certificate through the UI

dmelliott0311, 2016-07-18:
Can you add a feature to enable a user to add a certificate to enable HTTPS connections to the Web UI? It feels a little insecure to log into the admin console using a password passed through HTTP.

stllunatic, 2016-07-27:
I wonder if you could change the HTTP Port in the GUI to 443 to help achieve this? I was thinking the same  @dmelliott0311 

>  It feels a little insecure to log into the admin console using a password passed through HTTP.

>

ccrisan, 2016-07-27:
I totally agree. However, the only way to go is to setup an HTTPS reverse proxy (such as nginx) on a separate machine. That server should handle all the SSL stuff. Changing the port to 443 is not enough, as the traffic would still be clear text HTTP.

blah4-5078, 2016-07-27:
Definitely feels unsafe...

closed 2017-06-01 by bortek