motioneye-project/motioneyeos #406: setting static IP doesn't work through the web interface

ivr588, 2016-07-20:
Hi,

Recently I've needed to set the IP statically, so I went through the interface to manually assign it. Saved the changes, it rebooted but still on the old (dhcp-assigned) IP. Re-tried with the same result.

Will the static IP be defined in the usual linux places? Was thinking of just adding it there.

thanks,

ccrisan, 2016-07-20:
The static IP configuration is defined as simple variables in `/data/etc/static_ip.conf`. Check out that file and see if your values are correctly set in there.

Also, make sure you're not using both ethernet and wifi connections at the same time. If that is the case, the first one will eat up the static IP and the second one will continue to use DHCP.

ivr588, 2016-07-20:
Hi,

Here's the network config: http://photo.zyxst.net/meye1.png
The file /data/etc/static_ip.conf was blank when I opened it. Does just the static IP go in there, or other stuff like netmask and router IP?

thanks,

ccrisan, 2016-07-20:
Everything goes there. Can you please attach your boot.log file? It's odd that the file is blank while the UI shows what it shows.

ivr588, 2016-07-20:
I think I've found the problem :grinning: 
Before I came to set the IP, the camera had been running for a while. It was full. I took out the microSD and transferred the files onto another linux system, then deleted the files on the sdcard. Booted it up and I thought all was well.

It seems the filesystem is reporting that it's full even when the offending files have been removed. I tried to edit the static file in vi but it seemed to save it (but wasn't) so tried to modify the file with echo and got a more useful diagnostic:

# echo "192.168.1.203" >>static_ip.conf 
-sh: echo: write error: No space left on device

..hmmm I deleted everything in storage...

# df -h
Filesystem                Size      Used Available Use% Mounted on
/dev/root               174.3M    138.1M     23.6M  85% /
devtmpfs                426.8M         0    426.8M   0% /dev
tmpfs                   431.0M     16.0K    431.0M   0% /dev/shm
tmpfs                   431.0M      2.8M    428.2M   1% /tmp
/dev/mmcblk0p1           19.9M      8.7M     11.2M  44% /boot
/dev/mmcblk0p3           28.2G     28.2G         0 100% /data
/dev/mmcblk0p3           28.2G     28.2G         0 100% /home/ftp/sdcard
/dev/mmcblk0p3           28.2G     28.2G         0 100% /home/ftp/storage

# du -sh /data/*
88.0K	/data/etc
460.3M	/data/log
16.0K	/data/lost+found
4.0K	        /data/media
5.4M         /data/output

# du -sh /home/*
5.4M	/home/ftp

# du -sh /home/ftp/*
5.4M	/home/ftp/sdcard
4.0K	/home/ftp/storage

...so I think a fsck is in order! sorry for the noise

ivr588, 2016-07-20:
hmm well that formatted oddly....!

closed 2016-07-20 by ivr588