motioneye-project/motioneyeos #407: Rebootloop missing ethernet Pi Zero

Jimbojay, 2016-07-21:
Dear reader,

just purchased the zero 1.3 and trying to setup over wifi considering there is no ethernet-port using version motioneyeos-raspberrypi-20160705.img.gz. Unfortunately the first boot halts due to lacking ethernet-connection and falls back into a reboot loop.

[image]

Do I need a usb-ethernet dongle to setup? Below are some more or less comparable issues.

Raspberry Pi Zero reboot loop. #313:
ccrisan commented on May 29: "The next version will have the up-to-date Raspbian kernel. Assuming the new Raspbian kernel supports your device, the next motionEyeOS release will support it as well."

I am assuming that I therefore have the correct version (20160705) in which this is integrated. As I am kind of a Noob considering raspberry, do I need to undertake some extra steps?

Raspberry Pi Zero #338:
ccrisan commented on Jun 17: "Place your wpa_supplicant.conf file on the boot partition"

I found out that this is a Linux-based file. Is there a similar resolution for Windows? I am using 8.1 pro.

Thanks in advance for your reply!

Imre Dekker

ccrisan, 2016-07-21:
You can use your Windows to access the boot partition and place the wpa_supplicant.conf file there.

Jimbojay, 2016-07-21:
Thanks for your quick reply. Could you please tell me where to find wpa_supplicant.conf?

Kind regards

stllunatic, 2016-07-21:
The Pi Zero doesn't have built in wifi so you will need a wifi dongle to
get it connected to your network.

<https://cloud.githubusercontent.com/assets/10573528/17020426/c95116f6-4f41-11e6-925a-1d069eed50a6.jpg>

<https://github.com/ccrisan/motioneyeos/issues/313>:

Jimbojay, 2016-07-21:
I have the raspberry-wifi dongle connected already, but the initial boot apparently requires an ethernet-connection (see screenshot).

stllunatic, 2016-07-21:
You need to update your wpa_supplicant.conf file.

It should be located here:
/data/etc/wpa_supplicant.conf

I was able to view the file from my Windows computer--
1.) opened Notepad
2.) file opened file location: wpa_supplicant.conf **be sure to click the drop-down 'all files'**

The file should look something like the below (I removed my SSID and password)

update_config=1
ctrl_interface=/var/run/wpa_supplicant

network={
    scan_ssid=1
    key_mgmt=WPA-PSK WPA-EAP
    psk="your password"
    ssid="The name of your WIFI"
}

You should be able to save over this file and then your Pi should be able to connect to your WIFI and startup once your password and SSID are updated.

Jimbojay, 2016-07-21:
I navigated to C:\data on my windows laptop but this folder is empty

Therefore I:
1) opened notepad
2) copied your example text
3) WPA-EAP was deleted as my interface indicates WPA-PSK
4) adjusted the password and ssid. I let the " remain, is this correct?
5) saved the file as "wpa_supplicant.conf" on the root of the SD-card on which i installed motioneyeos
6) adjusted the automatic .txt extension to the required .conf extension
7) placed the SD-card into the Pi zero
8) Power up
9) Issue remains :(....

ccrisan, 2016-07-21:
@Jimbojay,

1. Write a *fresh* image onto your SD card. Do not try this with your existing installation.
2. Plug your SD card into your Windows machine.
3. Navigate with your Windows machine to your *boot* partition. It's the only partition your Windows machine will recognize.
4. Create the file called `wpa_supplicant.conf` and, using notepad, edit its contents as indicated above.
5. Boot your Pi with the SD card.

Jimbojay, 2016-07-21:
We are almost getting there!

@ccrisan due to your procedure the 'new first boot' tried to run wpa_supplicant.conf, but failed.

"Starting wpa_supplicant: failed"

- - - - - - - - - 
update_config=1
ctrl_interface=/var/run/wpa_supplicant

network={
scan_ssid=1
key_mgmt=WPA-PSK
psk= "my password"
ssid= "my network name"
}
- - - - - -

Thanks for all your support guys!

Jimbojay, 2016-07-21:
Do I perhaps need to add :

**ctrl_interface_group=wheel**

ccrisan, 2016-07-21:
You don't need that. You must not use that as the group `wheel` does not exist in motionEyeOS. Make sure your ssid and psk are correct.

Jimbojay, 2016-07-21:
psk and ssid are triple-checked and ok.

do I need to delete the parentheses (")?

ccrisan, 2016-07-21:
There are no parentheses in the sample wpa_supplicant.conf above.

Jimbojay, 2016-07-21:
excuse me I ment the    --->    "     "

ccrisan, 2016-07-21:
Nope. Leave them in place.

Jimbojay, 2016-07-21:
Any other ideas? I have a TP-Link extender in place which takes over the ssid-name, can that pose a problem? 

My other devices did not experience any problems when I installed the TP-Link extender.

ccrisan, 2016-07-21:
Make sure it works with Raspbian. You'll find plenty of resources online with regards to how to make Raspberry PI model A boards (and Zero) connect to a WiFi network. The same `wpa_supplicant.conf` should work with motionEyeOS as well.

Jimbojay, 2016-07-22:
I will try today and report the outcome.

Jimbojay, 2016-07-23:
This is not the case... 
1) I wrote Raspbian to SD
2) Booted Raspbian
3) Setup Wifi
4) Wifi tested by browsing the internet --> WORKED
5) looked at its wpa_supplicant.conf and copied it by hand into a file wpa_supplicant.conf (on laptop)
6) Wrote motioneyeos on SD
7) Copied the self-made wpa_supplicant.conf (from step 3)
8) Booted motioneyeos
7) Starting wpa_supplicant: failed

ccrisan, 2016-07-23:
What wifi adapter do you use? What version of motionEyeOS have you tried?

Jimbojay, 2016-08-01:
Sorry for the late update, but I wanted to test everything.

I figured out that it was the "USB OTG to USB shim" that interfered. When replaced with the cable-version: "USB OTG adapter kit", the problem was solved. Maybe the hardware was damaged somehow.

Thanks for your help!

closed 2016-08-01 by Jimbojay

jensenswitch, 2020-08-09:
I'm having the exact same problem, zero w but it looks for wired connection and panics when it cant find one..

starbasessd, 2020-08-09:
you have to place aq valid wpa_supplicant.conf file in /boot, or use the
os.conf file in /boot with OS_NETWORKLESS=true
to allow boot and then use a keyboard to create the wpa_supplicant.conf file

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?