motioneye-project/motioneyeos #409: using phone cameras with motioneyeos

ivr588, 2016-07-21:
Hi,

A while ago, I was able to access phone cameras repurposed as network cameras via an android software called IP Camera. I'd allocate the cameras a static IP and then it to the "add camera" dialogue on motioneyeos. I *think* it was motioneye on a linux machine rather than motioneyeos on a rpi2 but it was a few months ago and I'm not sure. I'm writing this now because on motioneyeos the cameras aren't recognised, even though one can browse to the camera from elsewhere on the LAN and see what the camera is seeing.

So I'm asking is motioneyeos different in this capability to motioneye?

thanks

ccrisan, 2016-07-21:
Nope. Not at all. You should be able to add network cameras to motionEyeOS without any problem. Just make sure you're using the correct MJPEG URL and not some URL that serves a web UI.

stllunatic, 2016-07-21:
At first I didn't think it worked either, but you have to use /video on the end for the MJPEG URL. 

I'm using a Galaxy S1 running the app "IP Webcam". The URL that worked for me was:
http://###.###.#.##:####/video

ivr588, 2016-07-25:
eventually half-remembered it 
Hopefully this will help other users with the same idea. If you're using IP Camera for Android, look at the 'advanced' section and in there it'll give all the urls for mpeg, static etc

closed 2016-07-25 by ivr588