motioneye-project/motioneyeos #415: Raspberry Pi Zero wpa_supplicant: failed (8192cu)

MichaelBachmeier, 2016-07-24:
Hi,

I've tried for hours to get the Zero run with the motioneyeos.
I've edited the wpa_supplicant.conf like this:

```
update_config=1
ctrl_interface=/var/run/wpa_supplicant
network={
    scan_ssid=1
    ssid="HelloWorld"
    psk="1234567890"
}
```
Then I get that error back:

```
Starting wpa_supplicant: failed (8192cu)
Configuring wireless network: no link
```

Changes at the SSID  have brought nothing. Same Issue.
In the first installtion there was not the "Starting wpa_supplicant: failed (8192cu)" error.
It was shown "dhcp" and then 5 times "wifi: disconnected".
But only one time and then the error above.

Raspbian works! Ideas?

Best regards

ccrisan, 2016-07-25:
Is this 20160705? If not, could you please try it with 20160705?

MichaelBachmeier, 2016-07-25:
Yes it is motioneyeos-raspberrypi-20160705.img.gz

MichaelBachmeier, 2016-07-25:
Sorry for double post. I use an EDIMAX EW-7811UN Wireless USB Adapter. But this works with raspbian

ccrisan, 2016-07-25:
Hmm I just tried the same hardware configuration (Pi Zero + Edimax 7811Un) and had no problem connecting to the WiFi. Can you please try to recover the /log/wpa_supplicant.log file from the data partition of your SD card and paste it here?

MichaelBachmeier, 2016-07-25:
I have now taken a brand new SD card. Same Issue!
I have no wpa_supplicant.log file and no log folder.

Here is my copy process:

`sudo bash writeimage.sh -d /dev/sdb -i "motioneyeos-raspberrypi-20160705.img" -n 'HelloWorld:1234567890'`

```
:: writing disk image to sdcard
220+0 records in
220+0 records out
230686720 bytes (231 MB, 220 MiB) copied, 24,9739 s, 9,2 MB/s
:: re-reading sdcard partition table
:: mounting sdcard
:: creating wireless configuration
:: unmounting sdcard
rmdir: failed to remove './.boot': Directory not empty
:: you can now remove the sdcard
```

ccrisan, 2016-07-26:
What do you mean you don't have a log folder? Are you sure you looked for it on the data partition?

MichaelBachmeier, 2016-07-26:
boot:

```
michael@michael-Linux /media/michael/F51C-6057 $ ls -la
total 8860
drwxr-xr-x  3 michael michael   16384 Jan  1  1970 .
drwxr-x---+ 7 root    root       4096 Jul 26 10:56 ..
-rw-r--r--  1 michael michael   13065 Jul  5 00:56 bcm2708-rpi-b.dtb
-rw-r--r--  1 michael michael   13328 Jul  5 00:56 bcm2708-rpi-b-plus.dtb
-rw-r--r--  1 michael michael   13101 Jul  5 00:56 bcm2708-rpi-cm.dtb
-rw-r--r--  1 michael michael   17932 Jul  5 00:56 bootcode.bin
-rw-r--r--  1 michael michael     144 Jul  5 00:56 cmdline.txt
-rw-r--r--  1 michael michael      45 Jul  5 00:56 config.txt
-rw-r--r--  1 michael michael    9716 Jul  5 00:56 fixup.dat
-rw-r--r--  1 michael michael  944857 Jul  5 00:56 fwupdater.gz
-rw-r--r--  1 michael michael 4134080 Jul  5 00:56 kernel.img
-rw-r--r--  1 michael michael 3889992 Jul  5 00:56 start.elf
drwxr-xr-x  2 michael michael    2048 Jul 26  2016 System Volume Information
-rw-r--r--  1 michael michael     136 Jul 25 22:11 wpa_supplicant.conf
```
data:

```
michael@michael-Linux /media/michael/159b92c2-7790-4f62-a99f-15529683d8f9 $ ls -la
total 34
drwxr-xr-x  20 root root  1024 Jul  4 21:54 .
drwxr-x---+  7 root root  4096 Jul 26 10:56 ..
drwxr-xr-x   2 root root  1024 Jul  4 21:54 bin
drwxr-xr-x   2 root root  1024 Okt  4  2015 boot
drwxr-xr-x   2 root root  1024 Jul  2 21:47 data
drwxr-xr-x   3 root root  1024 Okt  4  2015 dev
drwxr-xr-x  12 root root  1024 Jul  4 21:54 etc
drwxr-xr-x   3 root root  1024 Okt  4  2015 home
drwxr-xr-x   5 root root  1024 Okt  4  2015 lib
lrwxrwxrwx   1 root root     3 Jul  2 20:53 lib32 -> lib
lrwxrwxrwx   1 root root    11 Jul  2 21:04 linuxrc -> bin/busybox
drwx------   2 root root 12288 Jul  4 21:56 lost+found
drwxr-xr-x   2 root root  1024 Okt  4  2015 media
drwxr-xr-x   2 root root  1024 Okt  4  2015 mnt
drwxr-xr-x   2 root root  1024 Okt  4  2015 opt
drwxr-xr-x   2 root root  1024 Okt  4  2015 proc
drwxr-xr-x   2 root root  1024 Nov  1  2015 root
lrwxrwxrwx   1 root root     3 Okt  4  2015 run -> tmp
drwxr-xr-x   2 root root  1024 Nov 22  2015 sbin
drwxr-xr-x   2 root root  1024 Okt  4  2015 sys
drwxr-xr-x   2 root root  1024 Okt  4  2015 tmp
drwxr-xr-x   7 root root  1024 Okt  4  2015 usr
drwxr-xr-x   3 root root  1024 Jul  4 21:54 var
```
`find . -name "wpa_supplicant.log"` shows nothing

ccrisan, 2016-07-26:
There should be a third partition, the biggest one.

MichaelBachmeier, 2016-07-26:
omg, sorry

wpa_supplicant.log

`Successfully initialized wpa_supplicant`
`rfkill: Cannot open RFKILL control device`
`ioctl[SIOCSIWAP]: Operation not permitted``

messages

```
Jan  1 00:00:24 meye-47d8e3cf auth.err passwd: /etc/passwd: Read-only file system
Jan  1 00:00:24 meye-47d8e3cf auth.err passwd: can't update password file /etc/passwd
Jan  1 00:00:24 meye-47d8e3cf auth.err passwd: /etc/passwd: Read-only file system
Jan  1 00:00:24 meye-47d8e3cf auth.err passwd: can't update password file /etc/passwd
Jan  1 00:00:57 meye-47d8e3cf user.notice ethernet: no network connection available
Jan  1 00:00:57 meye-47d8e3cf daemon.info init: starting pid 835, tty '': '/etc/init.d/rcK'
```
dmesg.log

```
---- booting motionEyeOS 20160705----
[Wed Dec 31 23:59:59 1969] Booting Linux on physical CPU 0x0
[Wed Dec 31 23:59:59 1969] Initializing cgroup subsys cpuset
[Wed Dec 31 23:59:59 1969] Initializing cgroup subsys cpu
[Wed Dec 31 23:59:59 1969] Initializing cgroup subsys cpuacct
[Wed Dec 31 23:59:59 1969] Linux version 4.4.12 (ccrisan@supermicro) (gcc version 4.9.3 (Buildroot 2016.02-gd95d2d7) ) #1 Sat Jul 2 22:46:08 EEST 2016
[Wed Dec 31 23:59:59 1969] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
[Wed Dec 31 23:59:59 1969] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
[Wed Dec 31 23:59:59 1969] Machine model: Raspberry Pi Zero Rev 1.3
[Wed Dec 31 23:59:59 1969] cma: Reserved 8 MiB at 0x17400000
[Wed Dec 31 23:59:59 1969] Memory policy: Data cache writeback
[Wed Dec 31 23:59:59 1969] On node 0 totalpages: 98304
[Wed Dec 31 23:59:59 1969] free_area_init_node: node 0, pgdat c08865a0, node_mem_map d7c8c700
[Wed Dec 31 23:59:59 1969]   Normal zone: 864 pages used for memmap
[Wed Dec 31 23:59:59 1969]   Normal zone: 0 pages reserved
[Wed Dec 31 23:59:59 1969]   Normal zone: 98304 pages, LIFO batch:31
[Wed Dec 31 23:59:59 1969] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[Wed Dec 31 23:59:59 1969] pcpu-alloc: [0] 0 
[Wed Dec 31 23:59:59 1969] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 97440
[Wed Dec 31 23:59:59 1969] Kernel command line: dma.dmachans=0x7f35 bcm2708_fb.fbwidth=1184 bcm2708_fb.fbheight=624 bcm2708.boardrev=0x900093 bcm2708.serial=0x47d8e3cf smsc95xx.macaddr=B8:27:EB:D8:E3:CF bcm2708_fb.fbswap=1 bcm2708.uart_clock=3000000 bcm2708.disk_led_gpio=47 vc_mem.mem_base=0x1ec00000 vc_mem.mem_size=0x20000000  dwc_otg.fiq_fix_enable=1 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 rootwait ro rootflags=noload panic=10 quiet loglevel=1 ipv6.disable=1
[Wed Dec 31 23:59:59 1969] PID hash table entries: 2048 (order: 1, 8192 bytes)
[Wed Dec 31 23:59:59 1969] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
[Wed Dec 31 23:59:59 1969] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
[Wed Dec 31 23:59:59 1969] Memory: 371552K/393216K available (6068K kernel code, 437K rwdata, 1844K rodata, 380K init, 727K bss, 13472K reserved, 8192K cma-reserved)
[Wed Dec 31 23:59:59 1969] Virtual kernel memory layout:
                               vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                               fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                               vmalloc : 0xd8800000 - 0xff800000   ( 624 MB)
                               lowmem  : 0xc0000000 - 0xd8000000   ( 384 MB)
                               modules : 0xbf000000 - 0xc0000000   (  16 MB)
                                 .text : 0xc0008000 - 0xc07c2390   (7913 kB)
                                 .init : 0xc07c3000 - 0xc0822000   ( 380 kB)
                                 .data : 0xc0822000 - 0xc088f670   ( 438 kB)
                                  .bss : 0xc088f670 - 0xc0945270   ( 727 kB)
[Wed Dec 31 23:59:59 1969] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[Wed Dec 31 23:59:59 1969] NR_IRQS:16 nr_irqs:16 16
[Wed Dec 31 23:59:59 1969] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483647500ns
[Wed Dec 31 23:59:59 1969] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275 ns
[Wed Dec 31 23:59:59 1969] bcm2835: system timer (irq = 27)
[Wed Dec 31 23:59:59 1969] Console: colour dummy device 80x30
[Wed Dec 31 23:59:59 1969] console [tty1] enabled
[Wed Dec 31 23:59:59 1969] Calibrating delay loop... 697.95 BogoMIPS (lpj=3489792)
[Wed Dec 31 23:59:59 1969] pid_max: default: 32768 minimum: 301
[Wed Dec 31 23:59:59 1969] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[Wed Dec 31 23:59:59 1969] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[Wed Dec 31 23:59:59 1969] Disabling cpuset control group subsystem
[Wed Dec 31 23:59:59 1969] Initializing cgroup subsys io
[Wed Dec 31 23:59:59 1969] Initializing cgroup subsys memory
[Wed Dec 31 23:59:59 1969] Initializing cgroup subsys devices
[... 160 lines cut ...]
 * Loading kernel modules: done
 * Setting hostname: done
 * Starting syslogd: done
 * Configuring CPU frequency: done
 * Starting watchdog: done
 * Starting brownout detector: done
 * Starting eudev: done
 * Setting root password: done
 * Starting wpa_supplicant: failed (8192cu)
 * Configuring wireless network: no link
 * Configuring wired network: no device
---- shutting down motionEyeOS 20160705 ----
 * Stopping motioneye: failed
 * Stopping smbd: failed
 * Stopping nmbd: failed
 * Stopping proftpd: failed
 * Stopping sshd: failed
 * Stopping crond: failed
 * Stopping date updater: done
 * Stopping network: done
 * Stopping wpa_supplicant: done
 * Stopping eudev: done
 * Stopping brownout detector: done
 * Stopping syslogd: done
```

ccrisan, 2016-07-26:
Damn! I'm out of ideas. Really.

MichaelBachmeier, 2016-07-27:
no like this

**H**ello**W**orld:1234567890 (PW: 16 characters, only numbers)

blah4-5078, 2016-07-27:
I got nothing.

MichaelBachmeier, 2016-07-28:
i changed the SSID from **H**ello**W**orld to **H**ellworld and to helloworld.
No change solved the issue.

i check later a new SanDisk SD-Card.

MichaelBachmeier, 2016-07-28:
Same Issue!

ccrisan, 2016-07-28:
What security/encryption type have you set for your wifi network?

MichaelBachmeier, 2016-07-28:
WPA + WPA2

ccrisan, 2016-07-29:
Well at this point the only thing that I can do is to offer to remotely debug your issue, if you can attach a USB-ethernet adapter to your Zero (in addition to the WiFi dongle, possibly using a USB hub) and temporarily enable port forwarding to it.

MichaelBachmeier, 2016-07-31:
The Issue is fixed.

apparently the power supply was too weak. I use now a powered USB hub for the wifi stick.

thanks for your help

closed 2016-07-31 by MichaelBachmeier