motioneye-project/motioneyeos #416: Upgrade to 201660605 not working

digitizing, 2016-07-24:
I have Motioneyeos version 20160109 on a RasberryPI2  and has been working great.  I wanted to upgrade to the latest software.  I used the gui to check for upgrade and it advises the latest version is 201660605 and I confirm ok to upgrade but get message "The update process has failed".  What should I do to figure out the problem?  I did do a reboot and started looking at log files but still cannot figure out what is wrong.

ccrisan, 2016-07-25:
Please try to recover your motioneye.log file and attach it here. It should contain all the info needed to identify the cause of your problem.

digitizing, 2016-07-25:
Here is the log file.  Appreciate your help with this.

[motioneye.txt](https://github.com/ccrisan/motioneyeos/files/382080/motioneye.txt)

ccrisan, 2016-07-25:
I'm sorry but your motioneye log stops some time in February this year. Could you please try and recover the entire log file?

digitizing, 2016-07-25:
I think my SD card got corrupted somehow.  I tried to SSH in to extract the log file but no luck...wont accept the password.  I have done this in the past with no issue.  I think I will create a new SD with the latest code to get over this issue.

closed 2017-06-01 by bortek