motioneye-project/motioneyeos #417: SSH not working - clean image 2016-07-05

jimdaway, 2016-07-25:
Using an RPi2, with an 8gb microSD, new write of 2016-07-05, everything works except SSH. When i use PuTTY and try to open a connection to the IP of the unit (192.168.0.135) on port 22, the ssh window opens with the solid green cursor but no acknowledgement is displayed, no username prompt; nothing.

As I said, everything else seems to work very well (ftp, dropbox uploading, etc...)
[messages.txt](https://github.com/ccrisan/motioneyeos/files/380614/messages.txt)
[motioneye.txt](https://github.com/ccrisan/motioneyeos/files/380617/motioneye.txt)
[motion.txt](https://github.com/ccrisan/motioneyeos/files/380615/motion.txt)
[boot.txt](https://github.com/ccrisan/motioneyeos/files/380616/boot.txt)

I'm sure I'm missing something simple...

ccrisan, 2016-07-25:
Hmm, it should work out-of-the-box, and your boot log doesn't show any errors. What seems strange though is the following line in your messages log:

    ssh error: Hm, kex protocol error: type 30 seq 1 [preauth]

Upon a simple online lookup I found the following link that may help you with this error: https://blog.nytsoi.net/2015/07/13/putty-kex-error.

jimdaway, 2016-07-25:
That was indeed the fix. I looked through those logs and didn't see that!!

Thanks so much for the clarity and help!
This item can be closed.

closed 2016-07-25 by jimdaway